sccn/eeglab #385: Plugin Error - EEGlab won't load

Jacsbill, 2021-09-22:
### Description

EEG lab had suddenly stopped loading, having been working only two days ago with not changes to my laptop. It is an error with plugin_getweb 
This issue has been reported here: Plugin parsing error #160, by @arnodelorme, but with no solution I can make out.

<br>
#### Steps to Reproduce
Running eeglab via command line (was working yesterday)

<br>
#### Expected behavior:

<br>
#### Actual behavior: Error below.

`Index exceeds the number of array elements (74).

Error in plugin_getweb (line 63)
    plugin(iRow).downloads = stats{2}(iRow);

Error in eeglab (line 914)
        [stats pluginTmp] = plugin_getweb('', pluginlist, 'newlist');`

debuggin plugin_getweb reports 

`K>> stats{1}

ans =

  75×1 cell array

    {'AAR'                                   }
    {'Adjust'                                }
    {'AMICA'                                 }
    {'MB'                                    }
    {'ANTeepimport'                          }
    {'KB'                                    }
    {'ARfitStudio'                           }
    {'automagic'                             }
    {'batch_context'                         }
    {'BCI2000import'                         }
    {'bci2000legacy'                         }
    {'BDFimport'                             }
    {'KB'                                    }
    {'BERGEN'                                }
    {'bids-matlab-tools'                     }
    {'KB'                                    }
    {'bids-validator'                        }
    {'MB'                                    }
    {'bioelectromag'                         }
    {'biopac'                                }
    {'Biosig'                                }
    {'Blinker'                               }
    {'bva-io'                                }
    {'KB'                                    }
    {'CIAC'                                  }
    {'KB'                                    }
    {'Cleanline'                             }
    {'KB'                                    }
    {'clean_rawdata'                         }
    {'MB'                                    }
    {'Cogniscan'                             }
    {'corrmap'                               }
    {'KB'                                    }
    {'countBlinks'                           }
    {'CSP'                                   }
    {'ctfimport'                             }
    {'cwleegfmri'                            }
    {'Darbeliai'                             }
    {'detect_spindles'                       }
    {'KB'                                    }
    {'dipfit'                                }
    {'EEG-Beats'                             }
    {'KB'                                    }
    {'EEGBrowser'                            }
    {'KB'                                    }
    {'eegplot_w'                             }
    {'eeg_toolbox'                           }
    {'egilegacy'                             }
    {'EMDLAB'                                }
    {'envtopoForContinuous'                  }
    {'ERPLAB'                                }
    {'erppeakinterval'                       }
    {'erpsource'                             }
    {'erpssimport'                           }
    {'EYE-EEG'                               }
    {'MB'                                    }
    {'Eyesubtract'                           }
    {'FAA'                                   }
    {'FASTER'                                }
    {'KB'                                    }
    {'FDA'                                   }
    {'Features:←'                           }
    {'-EPCOH Analysis ←'                    }
    {'-- Functional Principal Component ←'  }
    {'Analysis (FPCA)←'                     }
    {'- ERP Analysis←'                      }
    {'-- Phase-Plane Plot/Derivative←'      }
    {'-- Functional Canonical Correlation ←'}
    {'Analysis←'                            }
    {'--- Other features:←'                 }
    {'- Smoothing with B-Spline/Fourier ←'  }
    {'basis functions. ←'                   }
    {'- Generalized Cross Validation ←'     }
    {'(GCV) for parameter estimation.←'     }
    {'- Functional Mean &amp; Variance'      }`

This seems to be same issue on previous post, but I don't know how to resolve. I am using a slightly older version of EEGlab, but I'm  nearing the end of a big project analyse and I'd rather not reinstall before finishing analysis of last few participants if I can help it.

<br>
#### Versions

| OS version     | Windows 10
| Matlab version | R2019b
| EEGLAB version |    2019.1 /  9.7.0.1216025 (R2019b) Update 1

DouweHorsthuis, 2021-09-22:
I seem to have the exact same issue using the same versions of everything @Jacsbill mentions
but when I load an older versions (v14_1_2b) EEGLAB works normally.

Jacsbill, 2021-09-22:
@DouweHorsthuis:- I've got my batch code temporarily working by running EEGlab from the root folder and manually adding path to the plugins I want before running code. It seems to work as long as I don't have anything that calls the eeglab function. A fix would be good though - I suspect this is an issue connecting to EEGLab server and might be a central bug. 

I've also pretty throughout checked for any updates in the last few days that might have triggered this and found nothing.

Jacsbill, 2021-09-23:
@DouweHorsthuis - I have done a quick fix on this. the function plugin_getweb is reading the first lines of the plugins. Basically, in the example above it should be reading after FDA. I think this is a bug, but as a quick fix to continue working I've just ammended line 43 of the plugin to not read past FDA

`  disp( [ 'Retreiving download statistics...' ] );
    [stats, status] = plugin_urlread('http://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_getcountall_nowiki.php');
    stats = textscan(stats, '%s%d%s%s%f%d%s%s%s%s%s%f', 'delimiter', char(9));
    for j=1:12;
        stats2{j}=stats{j}(1:61)
    end
    stats=stats2;`

with stats redefined my batch scripts all work fine. A temporary fix  as the size of stats may differ over time, but think this is a server bug.

DouweHorsthuis, 2021-09-23:
@Jacsbill Thanks for the solution. I moved to the 2021 version (it was time for me to update anyway).

Jacsbill, 2021-09-23:
@DouweHorsthuis is it all working with 2021? I tried 18 on another laptop and had same issue. I'll upgrade when I've finished pre-processing for this project.

DouweHorsthuis, 2021-09-23:
@Jacsbill yes, no problems for V2021.1 and no problems for v14_1_2b

arnodelorme, 2021-09-23:
OK, thanks. I am going to close the issue. Feel free to reopen. Cheers

closed 2021-09-23 by arnodelorme