sccn/eeglab #386: issue with plugin_getweb

yunshufan, 2021-09-23:
### when I type "eeglab" to load the GUI, I got the following error message:
[image]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 10] |
| Matlab version | [2021a and 2019b] |
| EEGLAB version | [eeglab2019_1] |

arnodelorme, 2021-09-23:
Are you using EEGLAB 2021.1?

@cll008 would you mind answering?

yunshufan, 2021-09-24:
I’m using eeglab2019_1

-Yunshu

arnodelorme, 2021-09-24:
Could you update to EEGLAB 2021.1 and try again?

Jacsbill, 2021-09-24:
@yunshufan see https://github.com/sccn/eeglab/issues/385 for a quick fix if you don't want to update just yet (worked for me to finish my current batch)

closed 2021-09-27 by cll008