sccn/eeglab #387: Extract epochs on selected data

diana-izdrui, 2021-09-27:
The EEGLAB toolbox cannot proceed to extract epochs on a selected data.

1. Select a specific range of data (using the Select data button)
2. Insert events every 0.1s (Edit -> Event values -> Insert events)
3. Extract epochs (Tools -> Extract epochs) -> the program cannot proceed to extract epochs to the selected data. It creates the epochs file, but with 0 Bytes and it returns to the first data imported (the entire signal) or at Import data pop_up. Also, in the Matlab Command window, it appears the following warning: 
'WARNING: The file size on disk does not correspond to the dataset, file has been truncated
pop_rmbase(): cannot remove baseline of an empty dataset'

| Matlab version: R2020a
| EEGLAB version | 2021

cll008, 2021-09-27:
Could you elaborate on some of your steps? e.g. in step 2 did you name your events? After step 2, do the events appear when you check the channel scroll plot? In step 3, how are you selecting the epochs?

You can also share your file at different stages here and I can take a closer look.

arnodelorme, 2021-09-27:
This is likely because you are trying to extract epochs which are too long.

diana-izdrui, 2021-09-28:
@cll008, please see below the extended steps:

1. Import data (already cleaned - 90s)
2. Edit -> Select data ->Time range [min max] (s) 3.137 5.642 -> Ok (epoch start 0.0s, epoch end 2.504)
3. Plot -> Channel data(scroll) - to check (ok)
4. Edit -> Event values -> Insert values -> Type 1 -> Latency from 0 to 2.5s, step 0.1 (26 values) -> Ok
5. Tools -> Extract epochs

@arnodelorme the events are each 0.1s, and the selected data is 2.5s. I proceed with the extract epochs steps previously on 634s, with events each 5s, and it worked, but it was not a selected data.

cll008, 2021-09-28:
Step 5. What are your epoch limits when you extract the epoch? If you are leaving it at the default [-1 2] seconds then this won't work because you only selected 2.504s of data...

closed 2021-09-28 by cll008

diana-izdrui, 2021-09-29:
Hello Clement,

Thank you for answering! I understand now. There is another question in my mind: How can I modify the epoch limits? I was unable to find how to modify it.

diana-izdrui, 2021-09-29:
@cll008 @arnodelorme Thank you for answering! I understand now where I've been doing it wrong.