sccn/eeglab #388: Plugin Error - EEGlab won't load

naloui1, 2021-09-27:
Hello,
I got the following errors when trying to load eeglab 2019_1 (by typing eeglab):

**Index exceeds the number of array elements (74).
Error in plugin_getweb (line 63)
    plugin(iRow).downloads = stats{2}(iRow);
Error in eeglab (line 914)
        pluginTmp = plugin_getweb('', pluginlist, 'newlist');**

I saw in  (https://github.com/sccn/eeglab/issues/385) that we can quickly fix  the problem without need to update a new version of eeglab by the modifying the line 43 of the plugin plugin_getweb as follows:

disp( [ 'Retreiving download statistics...' ] ); [stats, status] = plugin_urlread('http://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_getcountall_nowiki.php'); stats = textscan(stats, '%s%d%s%s%f%d%s%s%s%s%s%f', 'delimiter', char(9)); for j=1:12; stats2{j}=stats{j}(1:61) end stats=stats2;

However, I didn't understand why these numbers of 12 and 61.

Could you help me please?

Best regards

cll008, 2021-09-27:
12 because there are 12 fields in stats2 that need to be replaced. 61 to trim all lines after FDA.
I'll talk to @arnodelorme about it later. Looks like the stats fields are shifted (either formatting probably or someone uploaded something with missing fields). If we can't track down the root of the problem, the truncation that @Jacsbill came up with is a good fix.

I'm going to link #386 here as well so we don't have to track 3 separate threads.

cll008, 2021-09-27:
Thanks everyone for reporting the issue. Turns out the FDA description was formatted in a way that the older versions of EEGLAB cannot parse. Should start working again once the server updates.

closed 2021-09-27 by cll008

ethanbb, 2021-10-04:
It looks like a similar issue is happening now with the EYE-EEG plugin (I think there are tabs in the description) - can that be fixed as well? Meanwhile, I will update manually.

cll008, 2021-10-04:
@ethanbb thanks for reporting. Is this also an EEGLAB 2019 issue?

ethanbb, 2021-10-04:
Yes, it happened with a 2019 version, but not with the latest 2021 version.

cll008, 2021-10-04:
Thanks for reporting (and doing it on one thread). Arno just fixed this now. Will keep at eye out for carriage returns in the future...

Jacsbill, 2021-10-05:
Just to note that the 61 value for the quick fix above changed with the current bug, so you have to check where the trim is needed in stats every time.   

I'll be updating to matlab 2021 now I've finished my current project preprocessing rather than relying on quick fix.

cll008, 2021-10-05:
Thanks for the note @Jacsbill. We're correcting the problems on the plug-in server itself (changing text description to remove special characters). The reason this isn't an issue in 2021 is because the script automatically handles this.