sccn/eeglab #389: ERSP plot mismatch in eeglab2021 vs eeglab14

jjurgiel, 2021-09-29:
### Description

I'm seemingly unable to produce correct study ERSP plots for IC clusters in eeglab2021 (unless I'm missing some newly added specifier), compared to those obtained usingeeglab14. Study is using a paired analysis (2 conditions per subject ID, 1 group). Data from each subject is from the same session (same ICs within subject).

eeglab14 dipole decomp:
[image]

eeglab cluster 6 ERSP:
[image]

eeglab2021 produces same dipole decomp:
[image]

but different (and oddly high magnitude scale) ERSP plot for same cluster in eeglab2021:
[image]

I've also used the eeglab2021 .icatimef files themselves to basically recreate the eeglab14 ersp plot using my own code which is why I didn't expect an issue with those specifically. 

<br>
#### Steps to Reproduce

1. Create study with N subjects, 2 conditions per subject, single group, same ICs across conditions within subject (same recording session)
2. Precompute ERSP, input argument 'baseline', [time0 time1] added to ERSP arguments, along with default arguments that are present in the text box
3. Precluster (dipole locations only), cluster (k-means), and plot ERSP for a given cluster

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Windows 10] |
| Matlab version | [2020b] |
| EEGLAB version | [EEGLAB2021] |

arnodelorme, 2021-10-04:
Would you mind to upload your study and we will investigate.

jjurgiel, 2021-10-05:
I uploaded the study file and subject files. I didn't upload the .icatimef files as they're quite large, but please let me know if you'd like those from my end. They're simply component ERSP files using  'baseline', [-3500 -2500], using all ICs, as I think is described in study history.

arnodelorme, 2021-10-05:
@cll008 would you mind to have a look?

jjurgiel, 2022-01-17:
Update: I figured out that the issue with the study plotting is that trial info from the individual EEG files was not being detected when extracting ersp from the .icatimef files, and thus only a single trial from each file was being used instead of all trials for average ERSP, giving seemingly extreme values. The reason was because I was epoching my files from -4s to 0s prior to the locked event, which for some reason messed up the epoch numbers (my EEG.epoch variable info for epoch #1 was empty, and the true epochs and events started at epoch #2). After epoching from -4s to 0.05s (to make sure the locked event didn't get accidentally dropped out), the issue disappeared and epoch info was in tact.

Upon loading the study which wasn't working, I realized that the warning: "std_maketrialinfo: not the same number of time-locking events as trials, trial info ignored" appeared, which I didn't think was related but is associated with the same issue.

I'm not sure if EEGLAB14 and EEGLAB2021 handle these epoch details differently (i.e., EEGLAB2021 needs trial info variable to pull data while EEGLAB14 doesn't, as the latter uses individual files), but that's where this issue came from. Perhaps a flag check for scenarios where EEGLAB only detects a single trial for the study ERSP calculation for a particular subject may help for troubleshooting.

So this issue is solved for myself but please let me know if you need any other details of how to possibly recreate the secondary issue of eeg epoch details not forming properly.

cll008, 2022-01-20:
Thanks for the follow up @jjurgiel. I actually no longer work at SCCN or UCSD, and I'm not sure if @arnodelorme has the bandwidth to implement this feature. If you end up implementing that flag check yourself, I'm sure a PR would be appreciated.

arnodelorme, 2022-02-24:
This is a long one. Will get to it eventually.

arnodelorme, 2022-03-07:
@amisepa if you have time, can you have a look.

amisepa, 2022-03-07:
Looks like this issue is solved: "So this plotting issue is solved for myself (and isn't an issue with the study plotting or ERSP itself)"

Do you want me to add the flag check? "Perhaps a flag check for scenarios where EEGLAB only detects a single trial for the study ERSP calculation for a particular subject may help for troubleshooting."

arnodelorme, 2022-03-07:
OK, thanks

arnodelorme, 2022-03-18:
Let close the bug then. Thanks Cedric.

closed 2022-03-18 by arnodelorme