sccn/eeglab #406: Error regarding index integers (must be positive/same type of integer)

jellevandereerden, 2021-11-05:
I have a EEG data sample which contains six channels. Two of them are scalp channels (Tp9 & Tp10), the other four are implanted electrodes which are located near the medial geniculate nucleus (MGB) of the patients brain during deep brain stimulation (DBS) surgery. The exact location is not known, but these four channels are only millimeters apart in order to extract accurate local field potentials (LFP). I added the EEG data as EDF file and the channel locations as LOC file. When I try to run ASR with the following code for example, I receive an error saying: 

**Input -->** 
EEG = eeg_checkset( EEG );
EEG = pop_clean_rawdata(EEG, 'FlatlineCriterion','off','ChannelCriterion','off','LineNoiseCriterion','off','Highpass','off','BurstCriterion',20,'WindowCriterion',0.25,'BurstRejection','on','Distance','Euclidian','WindowCriterionTolerances',[-Inf 7] );
[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 1,'overwrite','on','gui','off');

**Output-->**
'Array indices must be positive integers or logical values.
  occurred in: 
    clean_windows: 105
    clean_asr: 146
    clean_artifacts: 233
    pop_clean_rawdata: 151'

or

'Integers can only be combined with integers of the same class, or scalar doubles.'

I tried to convert the EEG.data to double in order to make positive integers or integers, but this does not seem to work.
I cannot go into the detailed functions to check where it goes wrong.
Can you please help me out?

Thank you in advance,
[AD stimulus AD elektrodes.zip](https://github.com/sccn/eeglab/files/7481996/AD.stimulus.AD.elektrodes.zip)

Sincerely,
Jelle van der Eerden

cll008, 2021-11-08:
Hi Jelle,

To debug, please `edit clean_windows` and put a breakpoint on line 105. This seems more to do with an error in defining window lengths than in the data type. Let me know what you find. When reading the documentation of clean_rawdata I also don't see any documented usage of `WindowCriterionTolerances` so I wonder if that causes an issue, maybe steming from [this line](https://github.com/sccn/clean_rawdata/blob/98733205d15df1bc2e25e7cf172fcec2c2f1850f/pop_clean_rawdata.m#L115)?

arnodelorme, 2021-11-09:
@chkothe there is an issue with clean_windows at very high sampling rate (60k). Would you mind to have a quick look?

arnodelorme, 2022-03-07:
Moved to https://github.com/sccn/clean_rawdata/issues/38

closed 2022-03-07 by arnodelorme