sccn/eeglab #407: Channel Locations not saved when using command line

ioalessandro, 2021-11-05:
### Description

When generating channel locations in EEGLAB I use the code `EEG.chanlocs = pop_chanedit(chanlocs);` to open up the dialogue to select the mapping and to confirm the process for all my 24 electrodes. I am getting the channel locations perfectly with this method. Now I want to automate this by setting the code to `EEG.chanlocs = pop_chanedit(chanlocs, 'ok', 'ok');`. This now leads EEGLAB to show me Channel Locations _No (Labels Only)_. 

Another minor bug is that the locations flip the axis, which I am correcting with the following code (see below).

<br>
#### Steps to Reproduce
```
[ALLEEG EEG CURRENTSET ALLCOM] = eeglab;
EEG = pop_importdata('dataformat','matlab','nbchan',24,'data','/Volumes/IO_ONE/RevGabor2021/output/matfiles/ICA/Test_ICA_P21_short.mat','setname','EEG_P21','srate',250,'pnts',0,'xmin',0);
[ALLEEG EEG CURRENTSET] = pop_newset(ALLEEG, EEG, 0,'gui','off');

EEG.chanlocs.labels = {EEG.data.chanlocs.labels};
EEG.chaninfo.labels = {EEG.data.chaninfo.labels};
EEG.shortData = EEG.data.shortData;
EEG.data = EEG.data.data;
chanlocs = struct('labels', EEG.chanlocs.labels);
eeglab redraw;
EEG.chanlocs = pop_chanedit(chanlocs);
% EEG.chanlocs = pop_chanedit(chanlocs, 'ok', 'ok');
eeglab redraw;
EEG = pop_chanedit(EEG, 'nosedir','+Y');
[ALLEEG EEG] = eeg_store(ALLEEG, EEG, CURRENTSET);

```

<br>
#### Expected behavior:
EEG Channel Locations in my Data Set.

<br>
#### Actual behavior:
No EEG Channel Locations when using the command line.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | macOS 11.2.3|
| Matlab version | [R2019b |
| EEGLAB version | 2021.0 |

cll008, 2021-11-08:
Not sure what 'ok', 'ok' does for `pop_chanedit`. Please use optional inputs readlocs or lookup locs ([load](https://github.com/sccn/eeglab/blob/a94d887f6b9c106314bfba1433f974912cfa0231/functions/popfunc/pop_chanedit.m#L102-L104) or [lookup](https://github.com/sccn/eeglab/blob/a94d887f6b9c106314bfba1433f974912cfa0231/functions/popfunc/pop_chanedit.m#L109-L110) respectively). With regards to your axis flipping, this is dependent on whether you use the rotated.CED file or the standard.CED or standard.ELC file from the dipfit template.

closed 2021-11-08 by cll008

arnodelorme, 2021-11-09:
Hi Alesandro, use the EEGLAB GUI then type "eegh" to see the pop_chanlocs history then copy to your script.

ioalessandro, 2021-11-14:
Hi both, thanks for getting back to me, and sorry for the late reply. 

@cll008 the 'ok', 'ok' is intended to just press 'ok' on the two GUI windows opened by `pop_chanedit`. Which, again, if I do this manually will lead to correct channel locations. 

@arnodelorme the eegh command only gives me the following `[ALLEEG EEG CURRENTSET] = eeg_store(ALLEEG, EEG, CURRENTSET);` which is quite uninformative in my understanding. 

So far I am still unable to automate this.

Thank you both for your support!

cll008, 2021-11-15:
@ioalessandro you might have missed the part where I linked the optional inputs 'load' and 'lookup' locs. Also for Arno's comment, you use `eegh` after you use the GUI to load or lookup electrode locations.

ioalessandro, 2021-11-17:
Hi @cll008, again thanks for your answer! I really appreciate it! I had some trouble deciphering your solution, but I got it running now. Thank you!