sccn/eeglab #409: using pop_newtimef on group levels

DouweHorsthuis, 2021-11-12:
Hi,
I was wondering if there is any guide or way to use pop_newtimef on group level. I tried to look through all the links and videos that go over this function and time/frequency analysis in general using this in detail, but for as far as I can tell it only focuses on individual .set files and never on multiple peoples .set files or something like that. I know you can run it on 2 different conditions to compare them, but we have more than 2 participants and we would like to see their average response, not the difference between participants,

Thank you so much for your time! 
Douwe

cll008, 2021-11-15:
Hi Douwe, 
Please see [this tutorial on precomputing STUDY measures](https://eeglab.org/tutorials/10_Group_analysis/study_data_visualization_tools.html#precomputing-other-measures). ERSP/ITC supports `newtimef` arguments.

closed 2021-11-15 by cll008

DouweHorsthuis, 2021-11-15:
@cll008  Thank you, didn't know that. That'll work perfect.