sccn/eeglab #410: Mask non-significant ERSP/ITC in STUDY plot

luizpalucci, 2021-11-13:
### I would like to plot in green color ERSP/ITC derived from STUDY those non-significant results in the figure, similar to in separate subject-level ERSP/ITC plot  

<br>
#### Steps to Reproduce

1. Load .study file
2. Study >> plot channel measures
3. Plot ITCs

<br>
#### Expected behavior:
Green plot in the regions of the figure (example All C3 + All CZ + All C4 ; All subjects)  where non-significant ERSP/ITC occurs

<br>
#### Actual behavior:
Values/color are plotted independent of statistical significance

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 10*] |
| Matlab version | [*R2019a*] |
| EEGLAB version | [*eeglab2019_0*] |

arnodelorme, 2021-11-15:
It is already possible to do if you select significance at the group level 

https://eeglab.org/tutorials/10_Group_analysis/study_statistics.html

closed 2021-11-15 by arnodelorme

luizpalucci, 2021-11-16:
Thanks Arno,
In fact I have attempted to do this and works well when compared two conditions - i.e. in the output of the comparison (3rd/right figure).
I'm wondering whether/how is it possible to plot green/non-significant also in the first two figures (1st/left and 2nd/middle)?
Similar to the following paper - https://www.mdpi.com/1424-8220/20/6/1722/htm (Figure 4 to 7)
Many thanks again
Luiz