sccn/eeglab #413: pop_runica() overwrites channel locations

ioalessandro, 2021-11-17:
### Description

I am using the command line function `EEG = pop_runica(EEG, 'icatype', 'runica', 'extended',1,'interrupt','on');` to run ICA on my data that includes channel locations. After this command, the locations are gone, since the output of the function does not seem to include them and overwrites my EEG variable. How do I change this behaviour to merely adding the ica output structures to the EEG structure?

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [macOS] |
| Matlab version | [R2019b] |
| EEGLAB version | [2021.0] |

cll008, 2021-11-18:
`runica` shouldn't modify channel locations in any way, and I couldn't reproduce your problem. Are you sure the `EEG` you are inputting has channel locations?

- Is the EEG.chanlocs field missing entirely? 
- Which fields in the structure are empty?
- Are you looking at the GUI to determine 'Yes' or 'No' channel locations? If you use the command line you also have to type `eeglab redraw` to refresh the GUI

ioalessandro, 2021-11-19:
Thanks for getting back to me!

- The EEG.chanlocs field is present, yet empty after `runica`
- check out the attached screenshots for a pre/post `runica` comparison of `EEG `and the GUI
- I am, it reports the structures content correctly with `yes` pre ICA and `no` post ICA

[image]

[image]

cll008, 2021-11-22:
Channel location says `yes` in the GUI for both images attached, but I see it's empty as you mentioned (in the variable explorer). I can't reproduce this problem though. Since you are mixing command line and GUI, it's possible there are some non-bug mistakes. 
Can you confirm isempty(EEG.chanlocs) immediately before you run `EEG = pop_runica(EEG, 'icatype', 'runica', 'extended',1,'interrupt','on')` and the same after?

arnodelorme, 2021-11-23:
Closing this for now, but feel free to reopen.

closed 2021-11-23 by arnodelorme