sccn/eeglab #417: EEGLAB error in function pop_loadbva()line 48:

hy19950429, 2021-12-08:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]
I tried to open a .mat file, and the error appeared.
EEGLAB error in function pop_loadbva()line 48:
Quote "Channels" that does not exist.
<br>
#### Steps to Reproduce

1. [*First Step*]Import data
2. [*Second Step*]Using EEG functions and plugins
3. [*and so on...*]Using Brain Vis. Anal. Matlab file

<br>
#### Expected behavior:
[*What you expected to happen*]
Open the .mat file

<br>
#### Actual behavior:
[*What actually happened*]
EEGLAB error in function pop_loadbva()line 48:

<br>
#### Versions
EEGLab v2021.1
MATLAB R2020b
Win10

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2022-01-31:
Could you share your file for testing purposes.

arnodelorme, 2022-02-24:
Closing this bug for now, but if you can share your file please reopen the bug.

closed 2022-02-24 by arnodelorme