sccn/eeglab #418: compiled version can load a BDF file but source not

pooriakhan, 2021-12-09:
### Description

Hi
I have my one BDF files when I try to load them in compiled version 2020.0 first a popup comes up "Load data using biosig" and I just press ok next anotherone comes "Dataset info --pop_newset()"  again I press ok and the file would be loaded

I want to use this in my Matlab source I saved the script and I run this line:
EEG = pop_biosig('renamed_channel.bdf', 'importevent','off','importannot','off');
it throw error In edfannot2evt 
here:
`TimeStamp(N,1) = datenum(HDR.T0) + t0/(24*60);`
t0 ic Nan
 
https://i.postimg.cc/B62Ng50c/Screenshot-from-2021-12-09-04-21-19.png
https://i.postimg.cc/CL5kMCQP/Screenshot-from-2021-12-09-04-21-32.png
https://i.postimg.cc/Pxdvg1DM/Screenshot-from-2021-12-09-04-31-20.png

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*ubuntu 20*] |
| Matlab version | [*fill 2020b*] |
| EEGLAB version | [*2021.1*] |

arnodelorme, 2022-01-31:
Sorry for the late answer. Would you mind to share your BDF file?

arnodelorme, 2022-02-24:
Closing this bug, but if you can share a data file, please reopen it.

closed 2022-02-24 by arnodelorme