sccn/eeglab #423: Issue with newtimeftrialbaseln.m

LauraKlatt, 2022-01-20:
### Description

I encounter a problem with the newtimeftrialbaseln() function when I want to load erspdata using std_erspplot().
 
Specifically, I created a STUDY design and precomputed ERSPs. In order to load the data to plot topographies, I then run the following lines of code:
 
```
EEG=ALLEEG(1);
electrodes = {EEG.chanlocs.labels};
[STUDY, erspdata, ersptimes, erspfreqs] = std_erspplot(STUDY,ALLEEG,'channels',electrodes, 'noplot', 'on');

``` 
And I get this error message:
 
Index in position 3 exceeds array bounds. Index must not exceed 60.
 
Error in newtimeftrialbaseln (line 99)
            mbase = mean(P(:,:,tmpbase,:),3);
 
Error in std_readdata>processtf (line 412)
        dataout = newtimeftrialbaseln(P, xvals, g);
 
Error in std_readdata>@(x)processtf(x,xvals,opt.datatype,opt.singletrials,params) (line 246)
            dataTmp{iSubj} = cellfun(@(x)processtf(x, xvals, opt.datatype, opt.singletrials, params), dataTmp{iSubj}, 'uniformoutput', false);
 
Error in std_readdata (line 246)
            dataTmp{iSubj} = cellfun(@(x)processtf(x, xvals, opt.datatype, opt.singletrials, params), dataTmp{iSubj}, 'uniformoutput', false);
 
Error in std_erspplot (line 250)
    [STUDY, allersp, alltimes, allfreqs, events, paramsersp] = std_readdata(STUDY, ALLEEG, 'channels', opt.channels, 'timerange', params.timerange, ...
 
I tracked the error down to newtimeftrialbaseln. According to the documentation of newtimeftrialbaseln the tf input data should have the following the dimensions:
[freqs x times x trials] or [channels x freqs x times x trials].
 
However, it seems like the variable that causes the code to crash (variable P, see line 99) has the wrong order of dimensions. In my case it is [freq x times x trials x channels].

<br>
#### Steps to Reproduce

1. [*First Step*]: Create a STUDY design
2. [*Second Step*]: Precompute ERSPs, using  'trialbase' =  'full' as input 
    I used the following parameters

```
  [STUDY, ALLEEG] = std_precomp(STUDY, ALLEEG, {},'ersp','on','erspparams',...
    {'cycles', [3 0.5], 'freqs', [4 30], 'nfreqs', 52, 'timesout', 300, ...
    'baseline', [-600 -400], 'basenorm', 'off', 'trialbase', 'full'} ,...
    'itc','on');
```

3. [*Third Step*]: load erspdata with std_erspplot(), using multiple channels as input (e.g., as follows)

```
 EEG=ALLEEG(1);
 electrodes = {EEG.chanlocs.labels};
 [STUDY, erspdata, ersptimes, erspfreqs] = std_erspplot(STUDY,ALLEEG,'channels',electrodes, 'noplot', 'on');

```
<br>
#### Expected behavior:
erspdata, ersptimes, erspfreqs should be returned as output variables

<br>
#### Actual behavior:
The code crashes when newtimeftrialbaseln.m is called

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Big Sur 11.6.2] |
| Matlab version | [R2021b] |
| EEGLAB version | [v2021.1] |

arnodelorme, 2022-02-11:
Dear Laura,

Can you share your data?

Arno

arnodelorme, 2022-02-24:
Hi Laura,

Please share your data and we will have a look.

Arno

arnodelorme, 2022-02-24:
Closing this bug for now, but if you can share your file please reopen the bug.

closed 2022-02-24 by arnodelorme

LauraKlatt, 2022-02-25:
Hi Arno, 
Sorry for the late reply! I just sent you an email with a link to download the data. 
Best, 
Laura