sccn/eeglab #426: problem while loading data in eeglab

evastone97, 2022-02-14:
I am trying to upload a struct (mat file) that has been extracted from someone else and I want to load the data and be able to use EEGLAB but while loading the data I got some warnings and errors.

[image]
 
so in the end I only get one channel from my data, the sampling rate.

[image]

Any feedback will be useful thank you in advance.

amisepa, 2022-02-23:
Hi,

Can you please share the file?

evastone97, 2022-02-23:
Hi, I have finally solved this issue. Thank you for your reply. Now I achieve to import everything also channel location etc, but the problem now is that I have to import also the events from the file that I have which is an array with the stimulus and responses. The problem here is that I tried to use the pop_select function but I have some errors there so if you can help me with that question it would be great!

amisepa, 2022-02-23:
Ok, just attach the file here and I'll have a better idea of what you are trying to do.

arnodelorme, 2022-02-24:
Thanks for looking into that Cedric.

evastone97, 2022-02-26:
I can try to explain here: so I have successfully uploaded all the data except the events. (The data were in field trip format before and I have used the fieldtrip2eeglab function to be able to transfer them into EEGLAB format).
[image]

The problem now is that for the events I have a matrix with the stimulus types, the participant response, and the response time. How can I pass the events that I have in this format into eeglab? Thank you in advance.

amisepa, 2022-03-03:
One nice way to deal with events is to use BIDS: https://eeglab.org/tutorials/04_Import/BIDS.html

Otherwise, you can convert them to EEGLAB format in a script. You need to have each event as a row in EEG.event, with its code name in EEG.event.type and its latency (i.e., timestamp in ms). 
When you are done, type EEG = eeg_checkset(EEG); to make sure the structure is compatible with EEGLAB and it should add the EEG.event.urevent column automatically. 

Here is an example, with conditions 1, 2, 4, 8 being each a stimulus category in an ERP study. Note you can also rename them with character strings if you want, with EEG = pop_selectevent(EEG, 'event',1,'renametype','condition1','deleteevents','off');

[image]

Hope this helps,
Cedric

arnodelorme, 2022-03-07:
@amisepa closing this bug report. Feel free to reopen if we are not complete.

closed 2022-03-07 by arnodelorme