sccn/eeglab #427: pop_studyerp error

carleycode, 2022-02-16:
Hello, I am following along the SCCN tutorials. Have been fine up until tutorial 10. I am using v2021.1 of EEGlab, MATLAB R2021b, Edition	Windows 10 Pro
Version	21H1
Installed on	‎20/‎05/‎2021
OS build	19043.1526
Experience	Windows Feature Experience Pack 120.2212.4170.0

[image]

To reproduce: (follow Quick STUDY creation guidelines using study5 data)
1. file->Create study ->simple ERP STUDY
2. enter 2 for number of conditions and 5 for number of subjects
3. press okay
4. Name the study
5. enter condition names
6. enter condition data sets (for each of the 5 subjects put the -s253 file on the left and the -s254 file on the right)
7. press okay and get and bug window is generated instead of Channel ERP figure and pop_chanplot()

carleycode, 2022-02-16:
Incase it's helpful

MATLAB Version: 9.11.0.1769968 (R2021b)
MATLAB License Number: 40636010
Operating System: Microsoft Windows 10 Pro Version 10.0 (Build 19043)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.11        (R2021b)
Computer Vision Toolbox                               Version 10.1        (R2021b)
Curve Fitting Toolbox                                 Version 3.6         (R2021b)
Data Acquisition Toolbox                              Version 4.4         (R2021b)
Deep Learning HDL Toolbox                             Version 1.2         (R2021b)
Deep Learning Toolbox                                 Version 14.3        (R2021b)
EEGLAB Toolbox to process EEG data                    Version -           see     
Image Acquisition Toolbox                             Version 6.5         (R2021b)
Image Processing Toolbox                              Version 11.4        (R2021b)
MATLAB Coder                                          Version 5.3         (R2021b)
MATLAB Compiler                                       Version 8.3         (R2021b)
MATLAB Compiler SDK                                   Version 6.11        (R2021b)
MATLAB Report Generator                               Version 5.11        (R2021b)
Signal Processing Toolbox                             Version 8.7         (R2021b)
Spreadsheet Link                                      Version 3.4.6       (R2021b)
Statistics and Machine Learning Toolbox               Version 12.2        (R2021b)
Vision HDL Toolbox                                    Version 2.4         (R2021b)
Wavelet Toolbox                                       Version 6.0         (R2021b)

carleycode, 2022-02-16:
Same error occurs in following 10c tutorial with the STERN STUDY at the point of Study->precompute channel measures, click the ERP checkbox and press OK.  (Same error as above: EEGLAB error in function std_precomp() at line 234: Undefined function 'gcp' for input arguments of type 'char'.)

arnodelorme, 2022-02-16:
This has been fixed in the EEGLAB version which will be released this week

carleycode, 2022-02-16:
Okay awesome, thank you so much.

carleycode, 2022-02-16:
I will be able to retest it for you :)

arnodelorme, 2022-02-17:
You can try it here ahead of the release.

https://sccn.ucsd.edu/eeglab/currentversion/eeglab2022.0rc.zip

carleycode, 2022-02-17:
Yip, tutorials 10a - 10c now work fine, no errors with pre-release version. Thanks so much.

carleycode, 2022-02-18:
Bug with plotting ERP image in pre-release version. Would you like me to open a specific bug report for that one or do you know about it?
[image]

arnodelorme, 2022-02-18:
Could you upload your data? I will change the title of the bug report. No need to create a new one.

arnodelorme, 2022-02-18:
OK, I will give it a shot. You mean you imported the data using the BIDS importer?

https://openneuro.org/datasets/ds002680/versions/1.2.0

carleycode, 2022-02-18:
I still have the zip file if you would like me to upload but it is rather large :)

arnodelorme, 2022-02-18:
Ah ok. Thank you.

Arno

carleycode, 2022-02-22:
Hi Arno, it doesn't seem to be possible to upload the zip file here. Probably best to click on the link at the beginning of the tutorial, unless there is somewhere else you would like me to upload the data to? 
Cheers
Carley

arnodelorme, 2022-02-22:
I will give it a shot. Which section exactly of the tutorial creates a problem for you?

arnodelorme, 2022-03-07:
I have tried the tutorial. I loaded the STUDY, precomputed ERP-image and plotted and it worked. What kind of problem are experiencing? Please reopen the bug when you respond. Thanks so much

closed 2022-03-07 by arnodelorme