sccn/eeglab #429: Error in function pop_resample at line 224, array indices must be positive integers or logical values

jellevandereerden, 2022-02-22:
I want to perform downsampling on a .set file which includes 21 EEG channels and 8 DBS LFP electrode channels. However, when I want to start my preprocessing pipeline by downsampling from 2048 Hz to 500 Hz it gives me an error that array indices must be positive integers or logical values. I have tried to run it through the interface, as well as through Matlab command window.

arnodelorme, 2022-02-24:
Would you mind sharing your data file?

arnodelorme, 2022-04-19:
Closing this report. Feel free to reopen.

Bysmuth, 2022-05-02:
I just encountered the same issue. In my case, at least, it appears to be due to code within pop_resample() that loops over iEvt for continuous data (line 224; I'm using EEGLAB 2021). The first event is a boundary event with latency 0.5, but the if/else/end statement that checks for this condition misses it because I have already run pop_editeventlist(), an ERPLAB function that changes the boundary event so that EEG.event(1).type==-99. (EEG.event(1).codelabel=='boundary', but the check misses that.) As a result, it moves to the else statement and then fails with the error message above.

Bysmuth, 2022-05-03:
Happy to share a dataset if needed but I can reproduce this with the sample dataset (see below). Just to be clear, I recognize that this is not a problem inherent to pop_resample(); the issue is that the ERPLAB function modifies the event list in a way that pop_resample() does not expect, which is what yields the error.

```
EEG = pop_loadset('filename','eeglab_data.set','filepath','/System/Volumes/Data/data/matlab/eeglab/sample_data/');

EEG = pop_select(EEG, 'notime', [0 1]);   % remove first 1s of data to create boundary event at latency=0.5

% resampling (at same .srate) works fine despite 1st event being a boundary event
EEG_original_events = pop_resample(EEG, 500);

% but if pop_editeventlist() is used to modify event list first, boundary events are no longer recognized
EEG_edited_events   = pop_editeventlist( EEG , 'List', '~/Desktop/eeglab_data_elist.txt', 'UpdateEEG', 'code');
EEG_edited_events   = pop_resample(EEG_edited_events, 500);   % throws an error
```

arnodelorme, 2022-05-06:
This is indeed a problem. For now you can simply remove the event

```
EEG_edited_events.event(1) = [];
EEG_edited_events   = pop_resample(EEG_edited_events, 500);   % throws an error
```

@ammsimmons, is the type -99 synonym of boundary for ERPLAB. We could have all functions handle these events as boundaries (for filtering, epoch extraction, etc...) although it will be some work.

ammsimmons, 2022-05-13:
Yes, -99 is a synonym for boundary once ERPLAB has edited the events. A potential workaround is to initially downsample right after loading in the raw data but that may not be easily doable for a variety of reasons.

arnodelorme, 2022-05-18:
If you pull the latest version of EEGLAB (from Github) and go to the EEGLAB Preferences, you can now select event -99 as boundary. You might have to restart processing your data. Closing this for now, but feel free to reopen.

closed 2022-05-18 by arnodelorme

arnodelorme, 2022-10-11:
Thanks. I will have a look.