sccn/eeglab #430: Pop_runica 'chanid' input

faimai00, 2022-02-23:
Hi all,

I am using Matlab_R2021a and eeglab2021.1. 
I have an EEG.data matrix including 21 EEG channels (1:21, :), and 1 ECG channel (22, :). 

I am trying to run pop_runica from the Editor Window instead of using the GUI. 
When using the GUI, I select Tools > Decompose data by ICA > runica and I select EEG as my channel types. This allow me to run ICA on EEG channels only (excluding my ECG channel).

However, the channel selection selection does not appear in the code when I call eegh. The 'chanid' specification from the pop_runica() function should allow to specify what channels I want to run the ICA on, but if I specify it, I get the following error: 

Output argument "sphere" (and maybe others) not assigned during call to "runica". 
Error in pop_runica (line 439)   
[EEG.icaweights,EEG.icasphere] = runica( tmpdata, 'lrate', 0.001,  g.options{:} );  

I am using the following code:
%% Run ICA
[EEG ALLEEG CURRENTSET] = eeg_retrieve(ALLEEG,1);
EEG = eeg_checkset( EEG );
EEG.eegrank = EEG.nbchan - size(EEG.etc.noiseDetection.interpolatedChannelNumbers, 2) - 2; % Nbchan minus interpolated channels, reference channel, and ECG. 
EEG = pop_runica(EEG, 'icatype', 'runica', 'extended',1, 'interrupt', 'on', 'chanid', 1:21, 'pca', EEG.eegrank);
[ALLEEG EEG] = eeg_store(ALLEEG, EEG, CURRENTSET);

I have also try to input the channel type, i.e. 'chanid', {'EEG'} but the same happens. 
The error does not appear when I run pop_runica without 'chanid' specification. 

I am unable to find any help on previous documentation so please would you mind advising on whether I am doing something wrong? 

Many thanks. 

Irene

arnodelorme, 2022-02-24:
The correct spelling is "chanind" not "chanid" (could not find any instance of "chanid" in the pop_runica function. Let us know if that works for you.

closed 2022-02-24 by arnodelorme

faimai00, 2022-02-25:
Oh so sorry I did not realise! Thank you very much.