sccn/eeglab #431: Importing events from a MATLAB array or text file- Index exceeds the number of array elements

Katerinaprm, 2022-02-24:
Hi all, 

I'm trying to import events from a MATLAB array or text file. 

My text file is composed of 5 columns (number type edftype latency urevent). 
In the 'import event info--pop _improvement window I browsed my file'; in the input field column I completed number type edftype latency urevent; number of file header lines I put 1; time: 1E-3; 

However, it appears **an error in function improvement, index exceeds the number of array elements (5).**

[image]
Do you have any idea why this is happens and how can I solve it? 
Thanks in advance, 
Katerina

arnodelorme, 2022-02-24:
Katerina, would you mind sharing your event text file. You can share it here (drag and drop).

arnodelorme, 2022-03-07:
Hi Katerina, closing this bug for now, but feel free to post the file and reopen. Cheers

closed 2022-03-07 by arnodelorme