sccn/eeglab #432: EEG checkset error

saiviswanth, 2022-02-25:
Hi all 

I have been trying to synchronise the gait cycle to the EEG signal where my raw EEG file(continuous data) is a  .vhdr extension file. When my section of code of synchronisation is completed for each trial(total no of trials - 4)  I am getting a eeg checkset error - which says that it does not have an epoch field in the event struct of EEG workspace variable. I went inside the event struct to find out there is no epoch field.

[image]

[image]

Thanks in advance for your help 
Sai viswanth

arnodelorme, 2022-02-25:
Is the data continuous or are there trials? If you artificially created trials, then the event structure needs an epoch field that is equal to 1, 2, 3 or 4 indicating which trial the event fall in.

closed 2022-02-25 by arnodelorme

saiviswanth, 2022-02-25:
The data is continuous and there are four trials - MLP,Stroop,uplift,Walk.
So  could i know how to create an epoch field which is equal to my trial
event(ie 1,2,3 and 4).

mumumu-tang, 2023-11-22:
Have you solved your problem？I meet the same problem. :(