sccn/eeglab #433: dimension misunderstanding in pop_resample function

CoSineZxc, 2022-02-26:
### Description

When I resample a EEG dataset into 100 Hz, which originally has 92 channels, 856 trials, 750 data points (sampling rate 250 Hz) in each trials, I use function EEG = pop_resample(EEG, 100);. It appears an error which shows in the picture. 

[image]

From my understanding, the left side should be 1-by-300-by-856, and the right side should be 300-by-856. The number 300 means (750/250)×100. And the reason why error shows 343 is that (856/250)×100=342.4, which means the script mistakes the amount of trials for the amount of time points in each trial. And as for the 750 shows in the right side, it is also the similar mistake on dimension misunderstanding.

When looking at the script, I think there are two places I cannot understand. The first one is line 348 and 349. Why using function length() of the matrix here? Function length() returns the length of the largest array dimension. So every time if the amount of trials is larger than the amount of time points in each trial, the program will always misunderstand the meaning of dimension. I think length() should be replace into size(). The second one is the input of function myresample(), I can only run this function smoothly if I use the transpose the EEG data. 

Following are original script and what I think it should be 
the first question(line 348): 
```matlab
        X = 0:length(data) - 1;
        newpnts  = ceil(length(data) * p / q);
        XX = (0:newpnts - 1) / (p / q);
```
I think it should be
```matlab
%         X = 0:length(data) - 1;
        X = 0:size(data,2) - 1;
%         newpnts  = ceil(length(data) * p / q);
        newpnts  = ceil(size(data,2) * p / q);
        XX = (0:newpnts - 1) / (p / q);

```
The second question(line 153): 
```matlab
for index1 = 1:size(EEG.data,1)
    fprintf('%d ', index1);	
    sigtmp = reshape(EEG.data(index1,:, :), oldpnts, EEG.trials);
```
I have to transpose it to make it work
```matlab
for index1 = 1:size(EEG.data,1)
    fprintf('%d ', index1);	
    sigtmp = reshape(EEG.data(index1,:, :), oldpnts, EEG.trials);
    sigtmp=transpose(sigtmp);
```
|   |   |
|----| ---- | 
| OS version     | Win10 |
| Matlab version | R2021a |
| EEGLAB version | v2021.1|

Sorry for wasting your time on such a question, I am looking forward to your reply.

arnodelorme, 2022-03-07:
Can you share your data. Do you have the signal processing toolbox in MATLAB?
Most likely, a function is being overloaded and this is why the error arises. Type which "resample" to know what resampling function is being used. It should be in the signal processing folder.

arnodelorme, 2022-03-18:
This was fixed in 9b6d2c268
This is because you are not using the signal processing function and you are using data epochs.
Thanks for reporting.

closed 2022-03-18 by arnodelorme