sccn/eeglab #434: EEGLAB error occurs when I try to create a simple ERP study (pic attached)

okn10, 2022-03-02:
### Description
[image]

Best way to fix this ?

arnodelorme, 2022-03-07:
This has been fixed in the latest EEGLAB release EEGLAB 2022.0. Which version are you using? Thanks

arnodelorme, 2022-03-07:
Please reopen the bug if you still have problems.

closed 2022-03-07 by arnodelorme

okn10, 2022-03-07:
Hello, where can I download eeg 2022.0, this site still gives the 2021 version as the latest download:
https://sccn.ucsd.edu/eeglab/downloadtoolbox.php

okn10, 2022-03-08:
> OK, let us know
> […](#)

Is it possible to get a link to the 2022 version ?

okn10, 2022-03-11:
I still also keep getting this error even though I've downloaded eeg202
[image]
2