sccn/eeglab #436: ERSP plotting error (matrix size error?)

myonetta, 2022-03-05:
In my STUDY set, I'm having a problem of some ERSPs can't  plot.

When I plotted the ERSP topography of all subjects, error was occurred with this figure (and in command window,  "tftopo (): tfdata columns must be a multiple of the length of times").
[image]

In addition, it can plot each subject's ERSP of each channel, but cannot draw topography (with an error, "Unable to perform assignment because the size of the left side is 100-by-41-by-50 and the size of the right side is 100-by-41-by-50-by-19. ").
[image]

I thought there was a problem with my EEG-data matrix, but I can't find a solution.

Could you let me know if I have any necessary data processing?
Thank you in advance.

arnodelorme, 2022-03-07:
Can you send us your study (even if it is large).

arnodelorme, 2022-03-07:
Most likely, this is because EEGLAB cannot find the data.

myonetta, 2022-03-08:
Thank you for your support.
I'll share the google-drive link because I can't upload files here (although .dat* and .irc* were too big to upload).
[https://drive.google.com/file/d/1IbYq2Mi4hMjat7q0bH2rRey72tTNDhCo/view?usp=sharing](url)

If you need all files, I'll send as split archives.

myonetta, 2022-03-18:
Add
I uploaded all the data of this study to the GITHUB repository and invited you.
I look forward to hearing from you.
https://github.com/myonetta/erpserror

arnodelorme, 2022-03-18:
I have fixed the issue. This is because you had sessions. 3520139f5
Please pull the latest changes from GitHub and recompute ERSP and ERPimage. It works for me on your data.

closed 2022-03-18 by arnodelorme

myonetta, 2022-03-22:
I appriciated your helping. already I've already performed recompute, and it worked fine.

Best wishes,

arnodelorme, 2022-03-22:
Good. Thank you for getting back