sccn/eeglab #439: eegplugin_pca doesnt work

evastone97, 2022-03-15:
I followed this lecture https://sccn.ucsd.edu/mediawiki/images/9/95/EEGLAB2018_scripting5.pdf
and also the documentation https://eeglab.org/tutorials/contribute/design_plugin.html for EEG lab function examples (the zip file that is provided there is not working but ok I save the Matlab code that is given below as the function).

Then I save this function inside the plugin folder but then when I start again the EEG lab the RUN PCA option doesn't do anything. 

Can you help me with that?
[image]

Thank you.

arnodelorme, 2022-03-18:
There are sometimes special characters when copying and pasting from PDFs.
I have just tried this and it works for me.

```
function vers = eegplugin_pca(fig, trystrs, catchstrs) 

vers = 'pca1.00';
% find tools menu
menu = findobj(fig, 'tag', 'tools');
% PCA command
cmd = [ '[~, EEG.icawinv] = runpca(EEG.data(:,:));' ]; 
cmd = [ cmd 'EEG.icaweights = pinv(EEG.icawinv);' ]; 
cmd = [ cmd 'EEG.icasphere = eye(EEG.nbchan); disp(''Done'');' ];

% create menu
uimenu( menu, 'Label', 'Run PCA', 'CallBack', cmd, 'separator', 'on');
```

closed 2022-03-18 by arnodelorme

evastone97, 2022-03-23:
Thank you for your reply!

I solved my issue.

Now I have another one regarding to preclustering. I have created a new issue: https://github.com/sccn/eeglab/issues/443

Thank you for the support and the answers.

Kind Regards,
Evangelia

arnodelorme, 2022-03-23:
OK, thank you Evangelica. I will look into it.

evastone97, 2022-03-23:
Thank you in advance!