sccn/eeglab #442: Problems with plotting channel ERSP analyses in Study

rrooyyccee, 2022-03-21:
Using Matlab R2021b/ EEGlab 2022.0 on Windows 10. 

Our lab has been having trouble plotting ERSP analysis at the channel level.  ERP plotting is working fine. 
The error produced is: EEGLAB error in function std_readfile90 at line 130/ Unable to perform assignment because the size of the left side is 26-by-55-by-78 and the size of the right side is 26 -55-67
Other errors come up as well. 

The sample study has 3 subjects. Each subject has 2 .set files for two different 'drug' sessions. I've tried to do ANOVA with this within-subjects factor. I've tried changing parameters such as the baseline, frequency, etc. But this does not seem to get around the error. I was able to get LIMO working using the open-neuro data. And previously, as of about 2 months ago, was able to get this 
ERSP function to work. 

Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

Created a Study with these 3 subjects. The data has had artifact components already removed, so am not doing any preclustering. Data seem uniform in terms of electrodes and epoch lengths when I look at the files individually.  I am able to test plot an ERSP. After precomputing, when I attempt to plot ERSP at the channel level, then the errors appear. 

I expect to see ERSP images of the two conditions and the statistical testing. Instead am getting the above error messages. 

Here is a link to the file:
https://www.dropbox.com/s/okmdwqqukslbbmg/Test%20Study%20for%20Limo%20-%20Copy.zip?dl=0

closed 2022-03-21 by arnodelorme

rrooyyccee, 2022-03-22:
Thank you Arno! 
I was able to download the revised Github code for std_readfile and replace the file in my EEGlab functions folder.
It seems to be working now with the test dataset. I was able to plot the statistical comparison of ERSP across three drug conditions. Now I'll turn my attention back to using LIMO eeglab.  Have a good night!

[image]

Sincerely,

Royce Lee

arnodelorme, 2022-03-22:
Thank you for the feedback

rrooyyccee, 2022-03-31:
matlab r2021b/ EEGlab 2022.0/ replaced the std_readfile as abovbe.

Arno: 
I seem to be getting the same, or a similar error again. using the same dataset on a different computer when attempting to plot channel ERSP. With any electrode chosen, I get: EEGLAB error in function std_readfile() at line 129:
Unable to perform assignment because the size of the left side is 26-by-92-by-156 and the size fo the right side is 26-92-78. 

Matlab reads out: 

pop_loadset(): loading file C:\Users\rlee1\Desktop\Test Study for Limo - Copy\S1\s1c1.set ...
pop_loadset(): loading file C:\Users\rlee1\Desktop\Test Study for Limo - Copy\S1\s1c2.set ...
pop_loadset(): loading file C:\Users\rlee1\Desktop\Test Study for Limo - Copy\S2\S2C1.set ...
pop_loadset(): loading file C:\Users\rlee1\Desktop\Test Study for Limo - Copy\S2\S2C2.set ...
pop_loadset(): loading file C:\Users\rlee1\Desktop\Test Study for Limo - Copy\S3\S3c1.set ...
pop_loadset(): loading file C:\Users\rlee1\Desktop\Test Study for Limo - Copy\S3\s3c2.set ...
Done.
Done.
Reading subjects' data or looking up measure values in EEGLAB cache
.Reading subjects' data or looking up measure values in EEGLAB cache
.Reading subjects' data or looking up measure values in EEGLAB cache
.Reading subjects' data or looking up measure values in EEGLAB cache
.Reading subjects' data or looking up measure values in EEGLAB cache
...and then the error code displays.

I am not sure if this is relevant, but when plotting an individual channels timefrequency data: I see this warning in Matlab:

Warning: Some versions of Matlab crash on this function. If this is
the case, simply comment the code line 1655-1673 in newtimef.m
which aims at "plotting marginal ERSP mean below ERSP image" 
> In newtimef>plottimef (line 1467)
In newtimef (line 1433)
In pop_newtimef (line 342)
In questdlg2 (line 97)
In errordlg2 (line 55)
In eeglab_error (line 73)
In pop_chanplot (line 524)
In inputgui (line 218)
In pop_chanplot (line 251) 

I geta similar error with this other dataset: 3 sessions, each session a different drug. 
https://uchicago.box.com/s/4e8pgydxk8is2o2qbf6bnqg58cm4nmts

I'm confused about this error and what is causing it to appear despite replacing the std_readfile function?

Thank you,

Royce Lee

rrooyyccee, 2022-04-01:
An update if this is helpful: If I enter the same data into an alternative study design in which the exact same files are entered into a between subject's study (no session number, treating each .set file as a different subject) then producing an ERSP plot and statistical comparison works out fine. So, that leaves a few possibilities. 

One is that I may be designing the study incorrectly? I don't think this is true. I tried setting up a between-subjects study where I did not enter in session number. This lead to the creation of a single erp file and ersp file for each subject. Unless these files are concatenated, it seems that is not the right result. But I may be wrong about this.

arnodelorme, 2022-04-01:
I am downloading the data and will have a look