sccn/eeglab #445: Picard ICA algorithm not running on EEGLAB v2021.0. in MATLAB R2021a

bloniaszp, 2022-03-27:
@agramfort 
I am conducting functional connectivity analysis (i.e., renormalized Partial Directed Coherence) on 128-channel EEG data. Picard **is** working well in the runica() function on both windows and mac in MATLAB 2017b in EEGLAB 2021. However, it will not run in the command line, on a SLURM server, or in the GUI on MATLAB 2021a in EEGLAB 2021. It keeps saying error in pop_runica(line 481) (attached), if you run it with either the 500 or 1000 for 'maxiter.' The error message is that it turns an error that says too _many_ inputs. If you remove 'maxiter' and just do pop_runica(EEG, 'icatype','picard') you get that there are too _few_ inputs. 

Any thoughts on this would be very helpful. Thanks!

[image]

agramfort, 2022-03-28:
hi @bloniaszp 

can you share a full code snippet and an EEG mat file to replicate this?

or maybe @arnodelorme has an immediate idea. I think he wrote this wrapper for the PICARD ICA solver.

bloniaszp, 2022-03-31:
@arnodelorme @agramfort 

Hi there, sorry for the delay. So to clarify, picard works on windows and mac on MATLAB 2017b. However, I do not have access at the present time to check if it works on 2021a on mac or windows. I'm trying to use picard on a high performance computing cluster that has both a GUI and non-GUI version of EEGLAB 2021 and MATLAB 2021. 
The version of picard that is running on the cluster (which turns the error) is here: eeglab2021.0/plugins/tmullen-sift_public_beta-c5404fa778c6/external/regu/regu/picard.m
This is the same path that works on 2017b on mac and windows. 
The code that I am running that turns the error is the same from Arnaud. Specifically:

EEG = pop_runica(EEG, 'icatype', 'picard', 'maxiter',500);
or
EEG = pop_runica(EEG, 'icatype', 'picard', 'maxiter',500);

Do you think there is any interaction that might cause this issue or do you believe it just might not work on the HPC? The error is the one from above, namely that there are too many inputs. This is the same error I get when picard is not installed properly, but there is no difference between the instillation on the HPC and the mac and windows machines.

bloniaszp, 2022-04-03:
I was able to resolve this issue. Even though only one EEGLAB directory was in the path, I found that I had to specifically mask the ~/rugu/ folder in another EEGLAB folder somewhere else on the directory. Thanks!

closed 2022-04-03 by bloniaszp