sccn/eeglab #446: .asc file load error

Flzjler, 2022-03-28:
### Description

I have a problem in reading .asc file that with channel location information.

When I try to read it using the EEGLAB GUI: 

<b>Edit -> Channel locations -> Select .asc file -> OK</b>

the EEGLAB told me that  “Index at position 1 is out of array bounds (cannot exceed 177)”. And I found that 177 is the number of lines in the .asc file.

Has anyone encountered the similar problem?

<br>

### Versions

|   |   |
|----| ---- | 
| OS version     | MacOS、Win10 |
| Matlab version | R2021a |
| EEGLAB version | v2021.1 |

Flzjler, 2022-03-30:
> This is because your file is not of a definite format. Type "help readlocs" to get the list of available formats.

Hello, I type "help readlocs" to get the list of available formats and find .asc is a definite format.

"help readlocs" shows:
'.asc', '.dat':     
                Neuroscan-.'asc' or '.dat' Cartesian polar coordinates text file.

arnodelorme, 2022-03-30:
ASC mean ASCII which means text. Your file is probably not a neuroscan file, is it?

Flzjler, 2022-03-30:
> ASC mean ASCII which means text. Your file is probably not a neuroscan file, is it?

My .asc file is exported from neuroscan. Its contents are as follows:

;SCAN 4.0 ASCII Channel Description File
;
;Line format:
;#<channel number> <label>
;or:
;<display page> <channel number> <pos x0> <pos y0> <pos x1> <pos y1>
;
;A semicolon at the first column indicates a comment.
;
#1    FP1
#2    FPZ
#3    FP2
#4    AF3
#5    AF4
#6    F7
#7    F5
#8    F3
#9    F1
#10   FZ
#11   F2
#12   F4
#13   F6
#14   F8
#15   FT7
#16   FC5
#17   FC3
#18   FC1
#19   FCZ
#20   FC2
#21   FC4
#22   FC6
#23   FT8
#24   T7
#25   C5
#26   C3
#27   C1
#28   CZ
#29   C2
#30   C4
#31   C6
#32   T8
#33   M1
#34   TP7
#35   CP5
#36   CP3
#37   CP1
#38   CPZ
#39   CP2
#40   CP4
#41   CP6
#42   TP8
#43   M2
#44   P7
#45   P5
#46   P3
#47   P1
#48   PZ
#49   P2
#50   P4
#51   P6
#52   P8
#53   PO7
#54   PO5
#55   PO3
#56   POZ
#57   PO4
#58   PO6
#59   PO8
#60   CB1
#61   O1
#62   OZ
#63   O2
#64   CB2
0   1       0.401    0.088    0.439    0.133
0   2       0.459    0.075    0.496    0.121
0   3       0.517    0.084    0.554    0.130
0   4       0.373    0.173    0.410    0.219
0   5       0.546    0.168    0.583    0.214
0   6       0.229    0.232    0.267    0.278
0   7       0.287    0.242    0.324    0.288
0   8       0.345    0.246    0.382    0.291
0   9       0.402    0.246    0.439    0.291
0   10      0.459    0.246    0.497    0.291
0   11      0.517    0.246    0.555    0.291
0   12      0.574    0.243    0.612    0.289
0   13      0.632    0.236    0.670    0.281
0   14      0.689    0.228    0.727    0.274
0   15      0.228    0.338    0.265    0.384
0   16      0.286    0.347    0.323    0.392
0   17      0.344    0.347    0.382    0.392
0   18      0.403    0.348    0.440    0.394
0   19      0.459    0.350    0.497    0.396
0   20      0.517    0.347    0.555    0.392
0   21      0.575    0.345    0.612    0.391
0   22      0.632    0.343    0.670    0.389
0   23      0.689    0.342    0.726    0.388
0   24      0.227    0.450    0.265    0.496
0   25      0.286    0.449    0.324    0.495
0   26      0.345    0.448    0.382    0.494
0   27      0.402    0.449    0.439    0.495
0   28      0.459    0.449    0.497    0.495
0   29      0.517    0.447    0.554    0.493
0   30      0.574    0.447    0.611    0.492
0   31      0.632    0.445    0.669    0.491
0   32      0.690    0.444    0.727    0.490
0   33      0.170    0.552    0.208    0.598
0   34      0.228    0.552    0.265    0.598
0   35      0.286    0.552    0.323    0.598
0   36      0.342    0.553    0.380    0.599
0   37      0.400    0.553    0.437    0.599
0   38      0.458    0.552    0.496    0.598
0   39      0.517    0.553    0.554    0.599
0   40      0.575    0.552    0.613    0.598
0   41      0.630    0.552    0.668    0.598
0   42      0.688    0.552    0.725    0.598
0   43      0.748    0.552    0.784    0.598
0   44      0.227    0.693    0.264    0.738
0   45      0.287    0.669    0.324    0.715
0   46      0.345    0.659    0.382    0.705
0   47      0.400    0.656    0.438    0.702
0   48      0.458    0.653    0.496    0.699
0   49      0.517    0.655    0.555    0.700
0   50      0.575    0.656    0.612    0.701
0   51      0.633    0.664    0.670    0.710
0   52      0.690    0.695    0.728    0.741
0   53      0.278    0.807    0.315    0.853
0   54      0.343    0.774    0.380    0.820
0   55      0.401    0.766    0.439    0.811
0   56      0.459    0.764    0.496    0.809
0   57      0.517    0.761    0.554    0.807
0   58      0.575    0.773    0.612    0.819
0   59      0.637    0.809    0.674    0.855
0   60      0.340    0.885    0.377    0.931
0   61      0.401    0.899    0.439    0.945
0   62      0.459    0.918    0.496    0.963
0   63      0.516    0.899    0.553    0.945
0   64      0.575    0.879    0.612    0.925
1   1       0.028    0.032    0.139    0.161
1   2       0.167    0.032    0.278    0.161
1   3       0.306    0.032    0.417    0.161
1   4       0.444    0.032    0.556    0.161
1   5       0.583    0.032    0.694    0.161
1   6       0.722    0.032    0.833    0.161
1   7       0.861    0.032    0.972    0.161
1   8       0.028    0.194    0.139    0.323
1   9       0.167    0.194    0.278    0.323
1   10      0.306    0.194    0.417    0.323
1   11      0.444    0.194    0.556    0.323
1   12      0.583    0.194    0.694    0.323
1   13      0.722    0.194    0.833    0.323
1   14      0.861    0.194    0.972    0.323
1   15      0.028    0.355    0.139    0.484
1   16      0.167    0.355    0.278    0.484
1   17      0.306    0.355    0.417    0.484
1   18      0.444    0.355    0.556    0.484
1   19      0.583    0.355    0.694    0.484
1   20      0.722    0.355    0.833    0.484
1   21      0.861    0.355    0.972    0.484
1   22      0.028    0.516    0.139    0.645
1   23      0.167    0.516    0.278    0.645
1   24      0.306    0.516    0.417    0.645
1   25      0.444    0.516    0.556    0.645
1   26      0.583    0.516    0.694    0.645
1   27      0.722    0.516    0.833    0.645
1   28      0.861    0.516    0.972    0.645
1   29      0.028    0.677    0.139    0.806
1   30      0.167    0.677    0.278    0.806
1   31      0.306    0.677    0.417    0.806
1   32      0.444    0.677    0.556    0.806
1   33      0.583    0.677    0.694    0.806
1   34      0.722    0.677    0.833    0.806
1   35      0.861    0.677    0.972    0.806
1   36      0.028    0.839    0.139    0.968
1   37      0.167    0.839    0.278    0.968
1   38      0.306    0.839    0.417    0.968
1   39      0.444    0.839    0.556    0.968
1   40      0.583    0.839    0.694    0.968

Flzjler, 2022-03-30:
> ASC mean ASCII which means text. Your file is probably not a neuroscan file, is it?

I debug the code, in file "readneurolocs.m", the code(on 72 line):
```
% find first numerical index
% --------------------------
index = 1;
while ischar( locs{index,1} )
    index = index + 1;
end
```
Seems to be stuck in an endless loop.

ajm12223, 2022-08-31:
> ### Description
> I have a problem in reading .asc file that with channel location information.
> 
> When I try to read it using the EEGLAB GUI:
> 
> Edit -> Channel locations -> Select .asc file -> OK
> 
> the EEGLAB told me that “Index at position 1 is out of array bounds (cannot exceed 177)”. And I found that 177 is the number of lines in the .asc file.
> 
> Has anyone encountered the similar problem?
> 
> 
> ### Versions
> OS version	MacOS、Win10
> Matlab version	R2021a
> EEGLAB version	v2021.1

We are currently having the same issue with our Neuroscan .asc file and was wondering if you were able to find a solution