sccn/eeglab #450: pop_runica batch processing

crisglav, 2022-04-04:
Dear developers,

I just want to suggest an improvement of pop_runica. I have noticed that when running pop_runica on the batch processing mode the data of each recording is not stored and cleared up from memory after each run. Therefore the data is kept on memory and that makes difficult to work with large datasets. I have seen than in other pop_xxx functions this is dealt with eeg_eval, but the use of eeg_eval in pop_runica seems slightly different. Could dataset storing after applying runica to individual recordings be implemented? I would be happy to help and contribute with my code here.

Best,
Cristina

crisglav, 2022-04-07:
Hi, I've looked the run_ica code in more detail and I think this problem will be solved if we add after this line
https://github.com/sccn/eeglab/blob/73f6a7f43fdf2d57c0306fb900b88401ff3851fc/functions/popfunc/pop_runica.m#L324
the following line:
`ALLEEG(dats{index}(idat)).data = 'in set file';`
At least it did whatever I wanted to and allows me to apply runica to larger datasets. Could you have a look at it? Thanks a lot.

arnodelorme, 2022-04-19:
Good point. I have implemented a similar fix in 456ef3732

closed 2022-04-19 by arnodelorme