sccn/eeglab #458: error in pop_clust() line 252

evastone97, 2022-04-14:
Hello,

After the precomputing measures and the pre-calculation of the pre-clustering matrix I tried to cluster with kmeans 10, but I got the following error.
[image]
The problem is that I don't know exactly what to change in pop_clustering function because in line 252 there is nothing.

Thank you in advance.

evastone97, 2022-04-14:
This is the part of pop_clust function, but I don't know what to change from here

[image]

arnodelorme, 2022-04-19:
You would need to share your whole study so we could have a look. Does not matter if it is gigabytes as long as you can find a way to give us a link.

arnodelorme, 2022-04-28:
Closing this for now. Feel free to reopen

closed 2022-04-28 by arnodelorme