sccn/eeglab #459: ITC and ITC phase plot

mses61322, 2022-04-19:
Hello,
I'm wondering why I get an ITC phase plot instead of an ITC plot after I plot the channel time frequency. (As below)
[image]
What can I do to get an ITC plot just like the tutorial gives? Or it's just a bug?

|   |   |
|----| ---- | 
| OS version     | Win11 |
| Matlab version | R2021b |
| EEGLAB version | v2021.1 |

arnodelorme, 2022-04-19:
There is an option from the command line to plot just the ERSP, the ITC or both. Also, I believe the interface has checkboxes allowing you to plot just the ERSP or the ITC.

closed 2022-04-19 by arnodelorme

mses61322, 2022-04-19:
Actually, I mean the eeglab website and other online sources indicating the graph should be "ITC" rather than "ITC phase" just like the picture down below.  
[image]
I followed the manual steps but got a different result with the graph showing "ITC phase". Please let me know if I've missed anything.

arnodelorme, 2022-04-19:
You have masked for significance and there is no ITC significant here, hence a green plot. This is the checkbox to toggle showing the phase.
[image]

mses61322, 2022-04-19:
So, the default setting is to print an ITC graph or an ITC phase graph? Or could it be manually manipulated?

I've tested both check and unchecked the checkbox to plot, but I still can't get an ITC graph to further analyze the data like down below. Shouldn't it be ITC?
[image]

arnodelorme, 2022-04-19:
There is no significance so the graph is the same in both cases.