sccn/eeglab #460: Error when Scaling components to RMS microvolts

eeguser, 2022-04-22:
Hello, 

During preprocessing, I noticed a possible error following ICA correction. 

The EEG channels are in the 20-30 micovolt range before ICA, and in the 5000 microvolt range following correction. 

I use the Sobi algorithm in the Picard plugin: 
<< pop_runica( EEG, 'icatype', 'sobi', 'extended', 1, 'chanind', 1:temprank);
<< pop_subcomps(EEG, components_array, 0);

temprank is the rank of the matrix. 

This happens for some, but not all datasets. It seems there is something I do not understand when the components are scaled according to RMS microvolt. 

Do you know how I could fix this issue?

arnodelorme, 2022-04-28:
This would only occur if somehow the inverse weight matrix is not inverse of the weight matrix.
I tried on the tutorial dataset

```
EEG = pop_runica( EEG, 'icatype', 'sobi');
round(EEG.icawinv*EEG.icaweights*EEG.icasphere)

1	0	0	0	0	0	0	0	0
0	1	0	0	0	0	0	0	0
0	0	1	0	0	0	0	0	0
0	0	0	1	0	0	0	0	0
0	0	0	0	1	0	0	0	0
0	0	0	0	0	1	0	0	0
0	0	0	0	0	0	1	0	0
0	0	0	0	0	0	0	1	0
0	0	0	0	0	0	0	0	1
```

This looks OK to me.

arnodelorme, 2022-04-28:
But feel free to share a dataset and I will have a look.

closed 2022-04-28 by arnodelorme

eeguser, 2022-10-11:
Hello,

Thank you for addressing this.

I run matlab on Linux, perhaps that somehow affects how it works.

I see that if I compute double(EEG.data) the rank changes to become what I expect. Whereas rank on single precision is completely off.

I summed the diagonal of the output from the code you sent. It’s all ones and sum to rank.

For a couple of datasets, the ICA components are many orders of magnitude larger. The diagonal are still ones and sum to rank for these datasets.

Standard preprocessing are carried out before ICA. Filtrering, downsampling etc. All using eeglab functions.

I unfortunately cannot share the datasets. These are stored in a secure system without extraction possibilities.

Have you encountered this issue before, with very large amplitude ICA components? I have never seen it and have been running ICA for years.

For one of the features I am analyzing, the y-scale does not matter. But for the power analysis part, it does.

Kind regards
Knut

Sent from ProtonMail for iOS

arnodelorme, 2022-10-13:
Yes, it is common for rank to be off. We have several rank function in runica and will add a new one.