sccn/eeglab #461: eeglab2fieldtrip error in line 120

evastone97, 2022-04-26:
Hello, I have a problem with the function eeglab2fieldtrip.

[image]

So I have my datasets for each subject that contain three different event types inside and I first apply ICA to the whole dataset using EEGLAb, and then I wanted to separate the three conditions into three different subfiles. set. 

I use this line of code, but I receive an error in line 120 of the eeglab2fieldtrip function and I don't know what to do.

Can you help me with that?
Thank you in advance!

arnodelorme, 2022-04-28:
Could I have a copy of your dataset. It might be that your dataset does not have any events.

evastone97, 2022-04-29:
Yes, I can set the .set file but is says here that is not supported

arnodelorme, 2022-05-03:
Problem fixed in e88339156
Simply pull the changes with recursive submodules

git pull --recurse-submodules

closed 2022-05-03 by arnodelorme