sccn/eeglab #462: Saving/Loading after `pop_reref` does not yield the same `EEG.icawinv` matrix

mscheltienne, 2022-04-27:
`pop_reref` impacts both the data and ICA matrix. After re-referencing to a CAR, if you save and reload the dataset, the field `icawinv` does not contain the same matrix.

Consider the example below using the `'eeglab2022.0/sample_data/eeglab_data.set'` sample dataset.

### Without CAR

```
% Load
file_dataset = 'eeglab2022.0/sample_data/eeglab_data.set';
EEG = pop_loadset(file_dataset);
EEG = eeg_checkset(EEG);

% Drop non-EEG channel and crop dataset
idx = eeg_chaninds(EEG, {'EOG1', 'EOG2'});
EEG = pop_select(EEG, 'nochannel', idx, 'time', [0, 10]);

% Run ICA
EEG = pop_runica(EEG, 'icatype', 'runica', 'extended', 1, 'interrupt', 'off');

% Save
pop_saveset(EEG, 'filename', 'sample-raw.set', 'savemode', 'onefile');

% Load
EEG_loaded = pop_loadset('sample-raw.set');
EEG_loaded = eeg_checkset(EEG_loaded);

% Retrieve ICAWINV
icawinv = EEG.icawinv;
icawinv_loaded = EEG_loaded.icawinv;

clear EEG;
clear EEG_loaded;
```

If we compare `icawinv`: `all(abs(icawinv(:) - icawinv_loaded(:)) <= 1e-8 + 1e-8 * abs(icawinv_loaded(:)))` -> `1`. Both are equal.

### With CAR

```
% Load
file_dataset = 'eeglab2022.0/sample_data/eeglab_data.set';
EEG_reref = pop_loadset(file_dataset);
EEG_reref = eeg_checkset(EEG_reref);

% Drop non-EEG channel and crop dataset
idx = eeg_chaninds(EEG_reref, {'EOG1', 'EOG2'});
EEG_reref = pop_select(EEG_reref, 'nochannel', idx, 'time', [0, 10]);

% Run ICA
EEG_reref = pop_runica(EEG_reref, 'icatype', 'runica', 'extended', 1, 'interrupt', 'off');

% Reref
EEG_reref = pop_reref(EEG_reref, [], 'exclude', setdiff(1:EEG_reref.nbchan, EEG_reref.icachansind));

% Save
pop_saveset(EEG_reref, 'filename', 'sample-reref-raw.set', 'savemode', 'onefile');

% Load
EEG_reref_loaded = pop_loadset('sample-reref-raw.set');
EEG_reref_loaded = eeg_checkset(EEG_reref_loaded);

% Retrieve ICAWINV
icawinv_reref = EEG_reref.icawinv;
icawinv_reref_loaded = EEG_reref_loaded.icawinv;
```

This time, the comparison returns False, and indeed, looking at the underlying arrays:

```
>> icawinv_reref(1:3, 1:3)

ans =

   -1.7648    1.1855   -0.7265
   -1.2189    1.9435    0.7728
   -1.4890    2.4871    0.5610

>> icawinv_reref_loaded(1:3, 1:3)

ans =

   -1.7766    1.2944   -0.5295
   -1.2761    1.6962    0.7516
   -1.5517    2.2109    0.5206
```

|   |   |
|----| ---- | 
| OS version     | Windows 10 |
| Matlab version | R2022a |
| EEGLAB version | 2022.0 |

Am I doing something wrong here?

mscheltienne, 2022-04-27:
And if you re-reference, you don't get any more `inv(EEG.icaweights * EEG.icasphere) == EEG.icawinv` which you have before.

arnodelorme, 2022-04-28:
Yes, this is on purpose. The inverse weight matrix is referenced in the same way as the data (and the weight matrix is recomputed accordingly).

closed 2022-04-28 by arnodelorme

mscheltienne, 2022-04-28:
I get that the re-referencing also applies to the ICA matrix. But then, if the `weight` matrix is recomputed accordingly, why is the equality between `icawinv` and `inv(icaweights * icasphere)` lost? And why is an I/O roundtrip further impacting those matrix? That does not seem right.

arnodelorme, 2022-04-28:
Yes, good point. I used the tutorial dataset 

```
EEG = pop_loadset('filename','eeglab_data_epochs_ica.set','filepath','/System/Volumes/Data/data/matlab/eeglab/sample_data/');
```

And referenced several times EEG = pop_reref( EEG, []); and EEG = pop_reref( EEG, 1);
The product stayed the same

```
round(EEG.icawinv*EEG.icaweights*EEG.icasphere)

ans = 

1	0	0	0	0	0	0	0	0
0	1	0	0	0	0	0	0	0
0	0	1	0	0	0	0	0	0
0	0	0	1	0	0	0	0	0
0	0	0	0	1	0	0	0	0
0	0	0	0	0	1	0	0	0
0	0	0	0	0	0	1	0	0
0	0	0	0	0	0	0	1	0
0	0	0	0	0	0	0	0	1

```

mscheltienne, 2022-04-28:
Using `eeglab_data.set` from the [sample dataset ](https://github.com/sccn/eeglab/tree/develop/sample_data)and this code, I re-tested the `pop_reref(EEG, [])` on my macOS with MATLAB r2021b, and I got the same result.

```
% Load
file_dataset = 'eeglab_data.set';
EEG = pop_loadset(file_dataset);
EEG = eeg_checkset(EEG);

% Drop non-EEG channel and crop dataset
idx = eeg_chaninds(EEG, {'EOG1', 'EOG2'});
EEG = pop_select(EEG, 'nochannel', idx, 'time', [0, 10]);

% Run ICA
EEG = pop_runica(EEG, 'icatype', 'runica', 'extended', 1, 'interrupt', 'off');
```

Before and after referencing, `round(EEG.icawinv*EEG.icaweights*EEG.icasphere)` does give me identity as you show. 
However, `icawinv` is not anymore equal to `inv(icaweights * icasphere)`. This last operation yields this warning:

```
Warning: Matrix is close to singular or badly scaled. Results may be inaccurate. RCOND =  6.034795e-19. 
```

And looks like:

[image]

mscheltienne, 2022-04-28:
And it still does not explain why the I/O roundtrip impacts those matrices.