sccn/eeglab #464: updatedipplot() error

sanshumba, 2022-05-04:
Hi, i am very new to EEGLAB, please bare with me if this error is a result of something i am doing. I have several EEG .edf all from different patients. I wanted to work out specific effective connectivity for each of the .edf files. I downloaded the SIFT plugin for EEGLAB for this purpose. I followed the tutorial exactly, with t exception that the .edf files use channels as apose to components. I followed the process of SIFT
1. Pre-processing
2. Model fitting and validation
3. Connectivity
Once this is done i go to connectivity and when i attempt to view the time frequency grid i get the following error

[image]

Does anyone have experience with this error? am i  doing something wrong in my process? 

thank you in advance for the assistance
<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2022-05-06:
This seems to indicate that there is no source in the plot you are trying to update. Maybe you click on a different figure when the plot was being performed. I am not sure. Let us know if you find the solution.

closed 2022-05-06 by arnodelorme

sanshumba, 2022-06-07:
Hi, after alot of reading and trial and error i figured out that the cause of this error was that the data i was using did not have an channel location information as well as there not being any epochs. After adding these it worked fine.

Thanks