sccn/eeglab #465: EEGLAB error in function std_preclust() line 270

evastone97, 2022-05-04:
Hello,

I have this study desing (are more subjects but here I just use one)
and I have three different conditions 
Since I want to do ICA, I first used all the preprocessed data, I added the channel locations and then I decomposed with ICA
[image]

The next step was to precompute the component measures and to build a pre-clustering array, but I got the following warning when I choose PCA clustering and then build the pre-clustering array. 

So since the warning mentioned that I can only use the dipole and scalp maps then I have added the following options there:

Once I press okay I got this EEGLAB error in function std_preclust() line 270
[image]

Do you know what might be the reason for that?

arnodelorme, 2022-05-06:
I have made a change in 78a9fcb67 which might fix the issue. However, if it does not, I would need to see your STUDY.

evastone97, 2022-05-06:
ok, I am gonna check it now and see if it works. Thank you!

arnodelorme, 2022-05-06:
Actually this is the change a4fcca6df3888f90838a3bf34551801d450639ce not the one above.

evastone97, 2022-05-07:
ok now it works

evastone97, 2022-05-07:
I still have the same error... Can I send you my file?

arnodelorme, 2022-05-07:
Yes, please

evastone97, 2022-05-07:
Here is the link: https://drive.google.com/drive/folders/1SrjI_SA7s_ZMTy2mxqsYXy-2siz6myke?usp=sharing

evastone97, 2022-05-09:
I realize that the same warning for the same function appears again when I try to precompute the time-based measures  (ERPs, and spectra) and with the command window to say : *Out of memory.*

[image]

I can only precompute the scalp maps as measures before clustering (maybe there is less space needed for those)

What would you suggest me to do?

Also if I run a simple ERP study with the datasets that I have I will be able to get those time-based measures (at least the ERPs) ?

arnodelorme, 2022-05-09:
For this would you need to get more RAM for your computer. How much memory do you have?

closed 2022-05-09 by arnodelorme

evastone97, 2022-05-09:
I have 16Gb, I think is more than enough that's why I was wondering ..
[image]

arnodelorme, 2022-10-11:
Yes, 16Gb should be enough unless your datasets are very large. Simply decrease the time and frequency resolution when computing ERSP.