sccn/eeglab #466: plotting error (matrix size error)

snztz, 2022-05-05:
In my STUDY set, I am having problems with both ERPs and spectra plotting.

For instance, it cannot draw the power spectra of all subjects (with an error, "Unable to perform assignment because the size of the left side is 0-by-15 and the size of the right side is 159-by-15-by-14.").

[image]

I thought there was a problem with my EEG.data matrix, but I can't find a solution. Actually dimension of data is 14x320x15 in set files. Number of trials and epochs is 15. Number of channels is 14. Number of time points is 320. Sampling rate is 128. EEG structure looks right to me but please could you let me know if I have any necessary data processing?

I am attaching the file set containing some of the data (for only two subjects).

Thank you in advance.

[set and study files.zip](https://github.com/sccn/eeglab/files/8630596/set.and.study.files.zip)

arnodelorme, 2022-05-06:
I loaded your study, precomputed ERPs and spectrum, and then plotted them without problem. From the data you sent, would you mind to send me the exact sequence of events to reproduce the problem?

snztz, 2022-05-23:
First of all, thank you for your quick response. I was using 2021.1v of EEGLAB. Now, I've tried it with the latest EEGLAB version and it worked. Thank you very much again.

arnodelorme, 2022-05-23:
Good to know. Thank you

closed 2022-05-25 by arnodelorme