sccn/eeglab #467: runica() function error in line 674

evastone97, 2022-05-05:
My study contains 18 subjects and , I have created a study using all the subjects and I have added the channel locations. My next step is to do ICA (note that my data are already preprocessed using Fieldtrp). 

I choose to decompose by ICA from the GUI but then it gave me that error regrading the space and the runica function. 

for the decomposition I did:  runica 'pca', 40 
[image]

How can I solve this, am I doing something wrong?

Thank you in advance and sorry for lots of questions.!

arnodelorme, 2022-05-06:
You do not have enough memory. Try reducing PCA further or using a different computer to run ICA. You can also try a different ICA algorithm. https://eeglab.org/tutorials/06_RejectArtifacts/RunICA.html

closed 2022-05-06 by arnodelorme

evastone97, 2022-05-07:
ok, I will try that, So the main problem is the RAM of my PC? I prefer to use this specific ICA algorithm but yes I can try different ones and see what I am getting.