sccn/eeglab #469: pop_importeventwarining()

evastone97, 2022-05-07:
[](url)Hello, 

I am a bit confused with the documentation regarding the epoch and event loading.

So my data are already epoched and I use the pop_select() function to find the epochs of the study but I have trouble on how can I load the events. 
(*Note that my data are already preprocessed through the field trips and what I have is the .mat files of the respondents. 
I have created set files with the channel location and the epochs but I am missing the events. When I try to load them with importevent function I got this warning.

[image]

Is there an easy and understandable way how to load or find events from epoched data?

The EEGLAB documentation is pretty clear in general and has lots of information, but in my case, I have a specific structure of data and I got lost in the tutorials and everything. 

Thank you for your time and your help!

evastone97, 2022-05-07:
how can I attach a .set file here?

arnodelorme, 2022-05-08:
Yes, you can import events of epochs, but as the message say, you need to be careful of what you do and check your events after you import them. I am not sure that the fieldtrip2eeglab function will convert data epochs to the EEGLAB format. The better alternative is to extract epochs in EEGLAB.

closed 2022-05-08 by arnodelorme

evastone97, 2022-05-09:
Ok, I see. Regarding the epoched data, do you recommend me to run ICA first and then extract my epochs, or the other way around? I am trying to do both so I can see the difference. (Note that my data are already epoched and I just have to split them into more epochs later for my study purposes).