sccn/eeglab #470: std_rmalldatafields() line 65

evastone97, 2022-05-08:
Hi again, 

I am trying to load 18subjects with epoched data and create a study with those and I got this error while loading the dataset. 

[image]

I did a trial with the first 3 subjects and it worked perfectly, but when I am adding the 18 subjects I have this error.

Is this something that has to do with epoch consistency? because I have different epoch numbers for each subject.

Thank you in advance.

arnodelorme, 2022-05-08:
Would you mind sharing your data?

evastone97, 2022-05-08:
Yes my data are in this link here: https://drive.google.com/drive/folders/16BT6nvwei_PXOtBVh3TdX7Tw2YfzqyXY?usp=sharing
I have added the first 6 subjects (I still have the same error)
Let me know if you can see the data.

evastone97, 2022-05-08:
Let me know if you can see my data 😊

evastone97, 2022-05-08:
I think I manage to do it
So I relaized that when I am creating a study (by selecting the second option for browsing datasets it works)

But if I am loading the sets and then creating a study by using all the loading datasets I got this error.

So I think it's fine now, but it shouldn't give me an error with any of the two ways of data loading.

arnodelorme, 2022-05-08:
Thank you. So I should just try to create a study with the 6 subjects right?

arnodelorme, 2022-05-08:
OK, found the problem and fixed it in 7ef2a0fa6
It was due to a change I made 2 days ago.

closed 2022-05-08 by arnodelorme

evastone97, 2022-05-08:
hm ok but why it's saying that there are datasets without the same ICA pairs? 
What I should do except download the edited function that you have fixed in the issue 7ef2a0f?

arnodelorme, 2022-05-08:
Yes, you just download the function.
If you have different ICA decompositions for the same subjects on the same day, you do get a warning msg. Make sure you have assigned them to different sessions (=days).

evastone97, 2022-05-08:
ok, what do you mean by assigning the ICA decompositions them into different days?

evastone97, 2022-05-09:
I think I found what was the problem and I cannot precompute all the measures that I want, correct me if I am wrong:

So I got this warning;
[image]
and then what I did is to see where are those NAN values. The figures below show the stract and all the fields
[image]

[image]

So I went to the STUDY.cluster() and I tried to fill up the empty spaces following the order of my datasets. (Do you think that is ok to do that or should I try to find the main issue on why my datasets are not loading ok?)
[image]

Then after fixing this I was able to precompute all the measures that I wanted and I could also make my clusters.

[image]

Thanks again for your effort and all your answers! I really appreciate that.

arnodelorme, 2022-05-09:
Yes, you are missing some ICA decomposition for some reasons.

arnodelorme, 2022-05-09:
In theory yes, but it is better to rerun ICA

evastone97, 2022-05-09:
Another quick question, for the preprocessing part we have used binica algorithm and later after the proprocessing runica() algorithm was used with dimensionality reduction pca40. So if I rerun ICA what algorithm should I use? Is there a way to see what is happening if we combine 2 ica algorithms together?