sccn/eeglab #471: std_preclust() at line 270

evastone97, 2022-05-10:
Hello again, 

This is the [study ](https://drive.google.com/drive/folders/1wuYYs8BrPcpVnPuzc9ubHrdpnhCuFuKT?usp=sharing)that I am using and I am getting this specific error when trying to precompute measures. 

[image]

What might be the reason for that, can I change something in the function? (I have already download the functions with the changes that you have made but I still have the error )

arnodelorme, 2022-05-10:
Can you share the whole STUDY (including the data)?

evastone97, 2022-05-10:
ok with all the files and precalculations?

evastone97, 2022-05-11:
somehow I have solved this

closed 2022-05-13 by evastone97