sccn/eeglab #473: Unable to read MAT-file (s1.icaerp) file might be corrupt error in std_readfile() line 217

evastone97, 2022-05-13:
Hello 

I am trying to precompute components for clustering and I got this error

[image]

Here is the line from the function:

  215-  warning('off', 'MATLAB:load:variableNotFound');
  216- if ~isempty(opt.dataindices)
  217-      fileData = load('-mat', [ fileBaseName fileExt ], chanList{:}, 'trialinfo', 'times', 'freqs', 'parameters', 'events', 'chanlocsforinterp', 'chanall');
218-    else fileData = load('-mat', [ fileBaseName fileExt ], 'trialinfo', 'times', 'freqs', 'parameters', 'events', 'chanlocsforinterp', 'chanall');
 219-   end

[image]

I have already precomputed the ERP components and I can see them in my folder the problem is that they cannot be read. 

Here is the folder with all the data: 
https://dtudk-my.sharepoint.com/:f:/g/personal/s205726_dtu_dk/EpOB8X0fMMhFplu2dEpeoEMB7DP5uPS4OqgM6Ab2Qo9Jxw?email=arnodelorme%40gmail.com&e=rFwIkN

evastone97, 2022-05-13:
Ok somehow I have solved the issue. I precompute channel measures and then I recompute again the component measures and then I could create clusters with k means. Is the precomputing of channel measures necessary in order to do the pre-component measures?

I also have one question related to the IC scalp maps for each condition, can I ask here or there is another forum for general questions? 

Kind Regards,
Eva

arnodelorme, 2022-05-18:
No you do not need to precompute channels to precompute component measures. You can ask questions on the eeglablist 

https://eeglab.org/others/EEGLAB_mailing_lists.html#the-eeglab-discussion-list-eeglablist

closed 2022-05-18 by arnodelorme