sccn/eeglab #475: error in eeg_interp() at line 67

evastone97, 2022-05-15:
Hello,

I loaded some datasets and tried to create a study and after the creation I choose to add channel locations and I got the following error (ALL of my datasets had the same channel location, I load my chan location file in each one separately)

[image]

Then when I am trying to Precompute channel measures I got this error.
[image]

Is it because I have loaded the channel location file independently and not to all the study together?

arnodelorme, 2022-05-16:
Maybe you forgot to import channel locations for one dataset. Or maybe you have already removed bad channels. Feel free to share your study.

evastone97, 2022-05-17:
Yes, I am using preprocessed data in where the bad channels have been removed, and then interpolation was applied.

I still have the error when trying to compute the channel measures.. don't know what I am doing wrong (all of my datasets have channel locations)

Here is the link to my study.
https://dtudk-my.sharepoint.com/:u:/g/personal/s205726_dtu_dk/EUoBpDEEB6ZHporsM6CpOT4Bo0ujSqq_sEAiml78H3VVqQ

Thank you in advance!

arnodelorme, 2022-05-18:
The link requires a password.

evastone97, 2022-05-18:
Here is the new link: https://drive.google.com/drive/folders/1P4Fimp_YAUZD06kY0R-k3Tn8mzcoJeuU?usp=sharing

sorry for the inconvenience

evastone97, 2022-05-18:
I have try to add the channel locations separately to each one of the dataset but it didn't work.. I don't know why but I have tried the same process before in another study and I never had this error with the eeg_interp function

evastone97, 2022-05-19:
I have just uploaded a new zip file. Let me know if this is working.

arnodelorme, 2022-05-19:
Still error with the file. I Downloaded the while file but it says it is empty when I tried to uncompress it.

evastone97, 2022-05-19:
Ok I am gonna try another way then, don’t know what went wrong.. I will back then in a while

arnodelorme, 2022-05-19:
After you upload it, try downloading it and uncompressing it. Cheers

evastone97, 2022-05-19:
This is the new link: its working at least for me hope it works. Let me know.

 https://drive.google.com/drive/folders/1YmqKpZl1uL2-xKYeFeCbFnisECx8aH6G?usp=sharing

Thank you in advance.
Kind Regards, 
Eva

evastone97, 2022-05-19:
For me It stacks here when I am trying to precompute channel or component measures
(for s9)
[image]

arnodelorme, 2022-05-20:
Got the data. You can see that datasets 48 and 49 have different channel labels. Why is that, I do not know.

arnodelorme, 2022-05-20:
[image]

arnodelorme, 2022-05-20:
Delete channel location and reload the file

```
for ichan = 1:72, ALLEEG(ichan).chanlocs = []; end
EEG = ALLEEG;
EEG=pop_chanedit(EEG, 'lookup','/System/Volumes/Data/data/matlab/eeglab/plugins/dipfit/standard_BEM/elec/standard_1005.elc','load',{'/Users/arno/Downloads/eeginterpol/128_channel_easycap.elp.elp','filetype','autodetect'});
```

I reloaded the file from the GUI

closed 2022-05-20 by arnodelorme

evastone97, 2022-05-20:
I haven't noticed the different channel labels, and indeed don't know why and how that happened.  
So should I run the part of the code you provided above while having the study loaded in the GUI?

arnodelorme, 2022-10-11:
Correct or restart from scratch. Best