sccn/eeglab #476: pop_loadstudy error()

evastone97, 2022-05-15:
Hello,

I have tried to save my study and then open it but I got this error for load study.
Do you know what might be the reason for that?

[image]

arnodelorme, 2022-05-16:
Would you mind sharing your study? Might be due to a recent change? I am able to load studies myself right now.

jiadeL, 2022-05-16:
Sorry, I can't. It says GitHub doesn't support edf file.

evastone97, 2022-05-16:
I have tried it now but again I have the same error. Do you want all the data or just the study .set file?

arnodelorme, 2022-05-16:
Yes, I need the whole thing.

evastone97, 2022-05-16:
Here is the link: https://dtudk-my.sharepoint.com/:u:/r/personal/s205726_dtu_dk/Documents/s1_A_2.zip?csf=1&web=1&e=cGsF7b
I gave access to you. Let me know if it works. Thank you in advance!

jiadeL, 2022-05-16:
> External Sender. Be aware of links, attachments and requests.

Sorry, it doesn't work.

evastone97, 2022-05-16:
you mean, the study or the link?

jiadeL, 2022-05-16:
The link. I open it, but the websit say this:

Det virkede ikke
Vi beklager, men ***@***.*** kan ikke findes i adresselisten dtudk-my.sharepoint.com. Prøv igen senere, mens vi forsøger at løse problemet automatisk for dig.

evastone97, 2022-05-16:
Yes, is because I gave access only to Prof. Delorme. Do you have the same error @jiadeL ?

jiadeL, 2022-05-16:
Yes

evastone97, 2022-05-16:
..I hope there is a solution on that. I am trying now to create again my datasets and create a new study.  Hope it works for you too!

evastone97, 2022-05-17:
@jiadeL  do you also have this error when trying to create a study? 

[image]
 I realized that I don't get this error when I am uploading my datasets manually. (but this takes time..)

evastone97, 2022-05-17:
OK I recreate again my datasets and I have added them manually and now the study is saved successfully. 
But when I tried to precompute channel measures I got this error:
[image]

I have added channel location to all the datasets at the same time so what might be wrong for that?

jiadeL, 2022-05-17:
Sorry, I don't meet the same issue as you. But I meet an issue about the size of the data is too big. Do you know any way to deal with it?

[image]<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F82998648%2F168758556-0120e20b-0a30-4c9e-8585-74fb45c49b5d.png&data=05%7C01%7Cjliang6%40tulane.edu%7Cb1b514876ec5454203f108da37da122d%7C9de9818325d94b139fc34de5489c1f3b%7C0%7C0%7C637883707020911654%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=p7ebqbhqvbYVX0TX5whymZ3Q2pSdJBxLPfXyVf0L6HM%3D&reserved=0>

arnodelorme, 2022-05-18:
First check which dataset is missing channel location by typing on the command line

ALLEEG.chanlocs

This will show you the channel location structure for all the datasets.
If you still have problem, upload the data to https://drive.google.com/drive/folders/1l-cg6EU0ajErW0TIb0JTeoScgJMT6TWC?usp=sharing
as the link you provide does not allow to download the data. Thank you

closed 2022-05-18 by arnodelorme