sccn/eeglab #477: close error () Unrecognized EEG field name EEG

evastone97, 2022-05-17:
Hello again,

I have created a new study and I tried to precompute component measures but I got this error.

[image]

I cannot even find a function called close..
Do you know what might be the reason for that?

arnodelorme, 2022-05-18:
Strange. Never seen that before. Can you upload your study at https://drive.google.com/drive/folders/1Ewm6HSsa-tlLM3v_3ck-sJezChM6ijvv?usp=sharing

evastone97, 2022-05-18:
I have uploaded the study in the file. Let me know if you need anything else.

arnodelorme, 2022-05-19:
Zip file is corrupted, unfortunately.

evastone97, 2022-05-19:
I will try to create a smaller study and send a link again

arnodelorme, 2022-05-25:
I have fixed the issue in 17de44248 I think. Closing this for now, but it did not fix it, please let me know.

arnodelorme, 2022-05-25:
And make sure to reopen issue otherwise I will not see it.

closed 2022-05-25 by arnodelorme