sccn/eeglab #480: Error in function std_readfile() at line 30

jeenalrambhia, 2022-05-19:
I am trying to plot channel measures in a study initially it worked fine but now i am facing this error. please guide me how to solve this.

"unable to perform assignment because the size of the left side is 0-by-119 and the size of the right side is 249-by-119"

arnodelorme, 2022-05-19:
Are you using the latest version of EEGLAB. Would you mind to share your study (including the data files?)?

jeenalrambhia, 2022-05-19:
Thanks for your reply. This worked for me yesterday. I am not sure why it
is showing an error today.
Study is to differentiate between the rest state and mathematical
calculation state the dataset is of physionet EEGMAT.
please find the link below for dataset.
https://physionet.org/content/eegmat/1.0.0/
I am using the files in EDF format.

also i am facing another error which is while building pre-cluster  which
says
"cannot process components from different ICA decomposition of the same
subject"
error occurred in function std_readdata() at line 183

I am new to this please guide me if i have done anything wrong.

Thanks
Jeenal

mplaweck, 2022-05-20:
For testing purposes, have you tried to eliminate the group designation? I have seen a similar error with a dataset with two sessions per subject

jeenalrambhia, 2022-05-21:
Thanks it works but still i get the other error "cannot process components from different ICA decomposition of the same
subject" while building pre cluster 
error occurred in function std_readdata() at line 183
can you tell where i am going wrong

arnodelorme, 2022-05-21:
Can you recompute ICA at the study level and make sure the checkbox “merge data from different dataset for a given subject” or something similar is selected?

jeenalrambhia, 2022-05-25:
If i create a study using only subjects and condition as mentioned in the tutorial it is giving me error which states
" pop_select error:undefined argument 'trials' "
(error occurred in function pop_select() at line 220)

arnodelorme, 2022-06-01:
- I downloaded all the data. 
- Imported them all.
- Assigned names and condition to them since they did not have any
```
for iS = 1:length(ALLEEG), ALLEEG(iS).subject = ALLEEG(iS).filename(8:9); ALLEEG(iS).condition = ALLEEG(iS).filename(11); end
```
- lookup channels
- precompute spectrum (no interpolation)
- plotted spectrum

Work for me.

arnodelorme, 2022-06-01:
[image]

arnodelorme, 2022-06-01:
I am going to close this for now. Make sure you have the latest EEGLAB version from Github. If you find another problem with this data, please let me know.

closed 2022-06-01 by arnodelorme