sccn/eeglab #481: std_precomp - recompute "off" error in calling std_ersp, std_erp

alenarto, 2022-05-19:
### Description

The std_precomp.m script has a 'recompute' flag that can be set to "off" when precomputing measures, thus allowing a user to add new subjects to a study and precompute for that subject ONLY, while leaving existing (i.e., other subjects') files alone. In EEGLAB 2022, the recompute "off" setting fails for std_erp and std_ersp. 

**Details - std_erp.**
line155 - The conditional "_if exist(filename) && strcmpi(g.recompute, 'off') && nargout > 0_" ensures that erp files are ALWAYS recomputed because nargout is always 0 for std_erp.m calls from within std_precomp.m. Removing the last (nargout>0) criterion causes a failure in completion of the script (loading up the files) so presumably this was an active choice to force the script to recompute the ERP files (which admittedly does not take a long time, but the behavior is incorrect and the time adds up for large samples).

**Details - std_ersp**
line 240 - The filename for the condition is set as "_filenameersp   = [ g.fileout '.datersp'  ]_;". So the subsequent conditional at line 249 "_if exist( filenameersp ) && strcmpi(g.recompute, 'off')_" looks for files ending with *datersp*. This is inappropriate for the newest version (2022) given that timef files are produced not ersp files.  Changing the suffix to 'dattimef' resolves this issue.

<br>
#### Steps to Reproduce

1. Create study with X subjects and precompute desired files.
2. Add new subject to study.
3. Run std_precomp with "recompute" flag set to "off". 

<br>
#### Expected behavior:
The script computed new erp/spec/ersp files for the new subject ONLY.

<br>
#### Actual behavior:
As expected, the script computes SPEC files for new subject ONLY. Not expected, the script **RE**computes erp and timef (formerly ersp) files for ALL subjects.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | HPC cluster Unix |
| Matlab version | 2020a |
| EEGLAB version | 2022 |

arnodelorme, 2022-05-25:
Thank you. This should be fixed in 3f42492c3. Let us know.

closed 2022-05-25 by arnodelorme

alenarto, 2022-06-02:
@arnodelorme confirming that all looks good!

arnodelorme, 2022-06-02:
Thanks @alenarto