sccn/eeglab #482: EEGLAb error in function pop_loadset at line 139:

Dalezd, 2022-05-20:
[image]

The file can't be loaded. I'm so sorry I haven't found solution of this bug.
The chip of my laptop is M1.
<br>
#### Versions
|   |   |
|----| ---- | 
| OS version     | [*macOS Monterey 12.4*] |
| Matlab version | [*2021a*] |
| EEGLAB version | [*eeglab 2020.0*] |

arnodelorme, 2022-05-20:
This is likely due to the special characters. Could you put it in a folder without special characters?

closed 2022-05-20 by arnodelorme

Dalezd, 2022-05-20:
> This is likely due to the special characters. Could you put it in a folder without special characters?
[image]
I've tried before.But the problem still exists.

mengtao150, 2024-04-24:
I have the same problem, when I load sub1_2 it shows error 139 as shown in how, when I load sub2_2 there is no such problem.
[image]