sccn/eeglab #483: ERPImage Study Error - processerpim at line 474

mplaweck, 2022-05-20:
### Description
Created a simple study from a complicated dataset for testing and development - 1 subject with 2 sessions
Precompute study parameters without problem (ERP, Spectra, and ERPImage).
Plotting for ERP and Spectra works without Issue. 
Attempting to plot ERPimage produces the error 

EEGLAB error in function processerpim() at line 474
Output argument "outsort" (and maybe others) not assigned during call to "erpimage".

[image]

I have loaded the 1 subject 2 session dataset at 
https://drive.google.com/file/d/1RjXBw_GK2KRyEik8yBGpOIwB5zk2Tzc_/view?usp=sharing

To reproduce,
Load study, precompute study parameters and attempt to plot ERPimage

I am now running the developer Version 2022.1b, which solved a prior error with std_readfile with multi-session subjects

| OS version     | Win 11|
| Matlab version | R2021a |
| EEGLAB version | 2022.1b|

arnodelorme, 2022-05-23:
OK thank you. Will look into it.

arnodelorme, 2022-05-25:
Works for me, unfortunately. I have removed the annoying warning message about the parallel toolbox.

arnodelorme, 2022-05-25:
[image]

arnodelorme, 2022-05-25:
- Loaded the study
- Precomputed ERPimage
- Plotted the first channel

Did the same with version 2022.0 on the web and it worked as well. Sorry I cannot help if I cannot reproduce the issue. Maybe you have a conflict between EEGLAB versions on your paths? Closing the report for now, but feel free to reopen if you think there is something else I can do.

closed 2022-05-25 by arnodelorme

mplaweck, 2022-05-31:
I was able to replicate the issue, and it may result from my ignorance.
-Load the study
-Precompute study parameters, **but include the removal of prelabeled artifactual components**
-Attempt to plot the first channel

mplaweck, 2022-05-31:
Thank you for your attention to this matter
I was able to replicate the issue and put directions on the thread. Forgive me I this email is duplicative, but I am new to GitHub and both did not know how to “reopen” the issue nor is you would receive the updates to a closed issue.
Thanks
    Marty