sccn/eeglab #486: EEGLAB error in function std_rmalldatafields_line 65

evastone97, 2022-05-26:
Hello I still have this error when creating a study desing by using all the loaded datasets.

[image]

I have downloaded the function that was corrected in: [78a9fcb]  (https://github.com/sccn/eeglab/commit/78a9fcb676199e6f3ec414703546244ddadf9adb) but I still have the problem..

arnodelorme, 2022-05-26:
Are you just using the GUI or custom script. If just using the GUI please upload your entire study at 

https://drive.google.com/drive/folders/1_utwKi7PBRP7u18_4lM5IJnKiRJGXb1D?usp=sharing

And instructions on how to reproduce the issue after loading the STUDY.

evastone97, 2022-05-26:
Yes I am using the GUI. I have uplaoded a part of the dataset with the steps that I am following in the folder that you share.
Thank you in advance, 
Eva.

arnodelorme, 2022-06-01:
Hi Eva, I believe this problem was fixed a couple of weeks ago. I downloaded your data and could not reproduce it. You can use the EEGLAB version on Github if you want (or wait one month or so for new release).

closed 2022-06-01 by arnodelorme