sccn/eeglab #487: std_precomp error

evastone97, 2022-05-26:
Hello, 

I have a question. So for my datasets I use the IC label and I flag some of the components.
You reccomend to remove those components in the study desing instead of manually fi we are going to make a study. In my case I want to make a study but due to some problems the option of removing ICA artifactual compoentns (the one that we have in the pre calculation of channel measures) it didnt work. So I decided to remove them manualy.

Since with both ways we are going to have this problem with non equal components between the datasets, how am I going to be able to cluster then the data with those components?

[image]

arnodelorme, 2022-05-26:
> I have a question. So for my datasets I use the IC label and I flag some of the components.
> You reccomend to remove those components in the study desing instead of manually fi we are going to make a [...]

Yes, this is fine.

> Since with both ways we are going to have this problem with non equal components between the datasets, how am I going to be able to cluster then the data with those components?

Yes, you will. Make sure to remove components subject wide. Meaning if you have different datasets for the same subjects, they need to have the same component removed.

evastone97, 2022-05-26:
> I have a question. So for my datasets I use the IC label and I flag some of the components. You reccomend to remove those components in the study desing instead of manually fi we are going to make a [...]

Ok, I see. But I have realized that I have different components removed from same subject but different condition (4 datasets -different conditions for each subject, with more or less same number of components removed but not exaclty the same) si that normal?

arnodelorme, 2022-05-26:
> Ok, I see. But I have realized that I have different components removed from same subject but different condition (4 datasets -different conditions for each subject, with more or less same number of components removed but not exaclty the same) si that normal?

You would need to merge the datasets before applying IC label

evastone97, 2022-05-26:
> Ok, I see. But I have realized that I have different components removed from same subject but different condition (4 datasets -different conditions for each subject, with more or less same number of [...]

I was thinking about that, and I also tried to label them before the separation, but I thought I will get even better clusters later if  I remove them when they are separated (For example I found more artifactual components when the datasets were splited) Do you think that the clustering later its gonna be good enough even if I apply IC Label in the inital datasets?

arnodelorme, 2022-05-26:
Clusters will be more stable if they are merged.

evastone97, 2022-05-26:
Perfect thank you I will close this issue, since I have all the answers! thanks a lot :)

evastone97, 2022-05-27:
Hello,

This is a follow up issue on the IClabeling question that I have asked above.

So after doing IClabel then I choose this options for the precomputing of channel measures
[image]

But it took some time

[image]

and at the end it shoud the message out of memory. 
*My hard disc has extra GB free so I dont know why that is the case.
Note that I did the same thing before but I apply IC label in the separated conditions so when the files where not merged and it worked perfecly but now no.

arnodelorme, 2022-06-01:
This is the RAM memory not hard disk memory. Sometimes some datasets are slightly larger.

closed 2022-06-01 by arnodelorme