sccn/eeglab #488: pop_fileio bug importing .fif eeg channel locations

enricovara, 2022-06-03:
### Description

loading eeg data as so:
`
EEG = pop_fileio(fullfile(inFolder,'s2_A__Ve2_0.fif'),'dataformat','auto');
`
results in a seemingly ok EEG struct that contains EEG.chanlocs with the complete list of channel names in the first column and 'eeg' for all channels in the second columns but [] in all other columns for all channels, instead of location values.

Opening the same file using mne python yields location data.

The s2_A__Ve2_0.fif file and expected location data .mat files are attached.

Thanks!

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Mac OS 11.6.2 (20G314)] |
| Matlab version | [R2019b] |
| EEGLAB version | [2022.0] |
| Fieldtrip-lite version | [v20220602] |
| Fileio version | [v20220602] |
| bva-io version | [v1.71] |
[s2_A__Ve2_0.fif.zip](https://github.com/sccn/eeglab/files/8833321/s2_A__Ve2_0.fif.zip)
[chanLocs-64.mat.zip](https://github.com/sccn/eeglab/files/8833328/chanLocs-64.mat.zip)
[original_chan_locs.zip](https://github.com/sccn/eeglab/files/8833335/original_chan_locs.zip)

arnodelorme, 2022-06-21:
Yes, it is strange. The nose is pointed in the wrong direction. I set the option in the EEGLAB channel editor to point the nose to the +Y direction. Yet, some channels are clearly misplaced (FP2 for example).

arnodelorme, 2022-06-21:
[image]

arnodelorme, 2022-07-13:
To do: import FIF file properly

arnodelorme, 2023-03-24:
I have imported other FIF files containing EEG and the orientation was OK. I have added a warning message. a9d7c4335

closed 2023-03-24 by arnodelorme